Ultimaker/CuraEngine #798: Mesh Type Modify settings Overrides Extruder Choice

AbeFM, 2018-06-19:
Model printed in one material (extruder #1), when using a modify-settings mesh, will print with the material of the modifying mesh's extruder selection instead of the original object.

**Application Version**
3.4b

**Platform**
win10

**Steps to Reproduce**
Load a model. Add a support blocker. Press "move", select blocker. Resize, reposition.
Select 'per model settings', choose mesh type "modify settings".
Selected "compensate...overlaps" (doubt this matters)

**Actual Results**

[image]

**Expected results**
Object should continue to print with its own extruder. If the mesh is to modify this, it should be a selection under modify settings.

**Additional Information**
You're wasting my glue, son!

ChrisTerBeke, 2018-06-19:
Although it's probably an engine issue, it's better to post this in the Cura repository as there is a lot more traffic there (also from the core devs).

diegopradogesto, 2018-06-20:
But maybe you want to modify a mesh by printing part of it with the other extruder, right? I don't think this is an issue. And you always have the extruder selector in the left to select the extruder that is used to print a mesh.

AbeFM, 2018-06-20:
The issue is the "print this modifier in Ext2" doesn't make a ton of sense, you're NOT printing that object. What you are printing is the object inside, and while  you COULD use it that way, it's counter-intuitive.

For that matter, if I WANTED to use it the way described, I would look for it under per-model-settings.

diegopradogesto, 2018-06-21:
Closing this in favor of https://github.com/Ultimaker/Cura/issues/3959

closed 2018-06-21 by diegopradogesto