Ultimaker/CuraEngine #799: [3.3.1] Experimental Bridging with Dual Extrusion

Galfinite, 2018-06-21:
**Application Version**
3.3.1

**Platform**
Windows, 64bit

**Display Driver**
N/A

**Steps to Reproduce**
Enable bridging with dual extruders.
I used the benchy dual extruder model files.
Export sliced gcode

**Actual Results**
First layer of second color (with Benchy, it's the deck layer) is considered to be a bridge due to no support from the model for the same extruder.

**Expected results**
First layer should be printed normally, since there is in fact no bridge due to being supported by the model for the other extruder.

**Additional Information**
Sliced gcode/stl's attached.

[3DBenchy Dualprint.zip](https://github.com/Ultimaker/CuraEngine/files/2124964/3DBenchy.Dualprint.zip)

smartavionics, 2018-06-22:
Thanks for the report, I will investigate.

smartavionics, 2018-06-22:
Could you please attach a project file as I am not familiar with setting up dual extruder prints, thanks.

smartavionics, 2018-06-22:
Don't worry about the project file as I have worked out how to slice dual extruder parts and have already fixed the problem. The change has been incorporated into a pending PR (#738). Thanks again for the report.

Here's the first deck layer no longer being printed as a bridge:

[image]

Here's the roof that is being printed as a bridge:

[image]

Galfinite, 2018-06-22:
Awesome, thanks for the quick response!

Vandrasc, 2018-07-06:
Thank you for reporting it, and thanks @smartavionics for fixing it! 

Cura-5240.

closed 2018-07-06 by Vandrasc