Ultimaker/CuraEngine #801: Feature Request: Add object/model/mesh name comments to gcode blocks

paukstelis, 2018-06-28:
This is a request to place a comment with object names at the beginning of each gcode block associated with an object. This would something along the lines of: `;OBJECT:<name or meshgroup index>` . This feature would allow streaming servers for printing (e.g. octoprint) to have control over individual objects during printing. Specifically, this feature would allow use of the [Octoprint-Cancelobject ](https://github.com/paukstelis/Octoprint-Cancelobject) plugin, which currently is limited to use with Simplify3D and Slic3r. If a text name is not doable, I would think even the mesh's meshgroup index would work to demarcate objects in the gcode.

paukstelis, 2018-06-29:
Just saw the feature_mark_models_in_gcode branch. Will check it out.

paukstelis, 2018-06-30:
Both branches that add the Mesh comments work nicely. How difficult would it be to pass the mesh's filename from the frontend to provide a slightly more descriptive comment? I mean, I know what `;MESH:75092768` means, but others might not :)

alekseisasin, 2018-07-13:
@Hi @paukstelis , the MESH:75092768 is a memory address of this object, which is generated by a system. For now, we did not define the constraint for manually ( via a front end) defining the object names. At this moment it is not in our plan, but we get requests for it.

paukstelis, 2018-07-13:
@alekseisasin, thank you for the reply. Forgive me for not fully understanding how this is all laid out, but how are meshes passed from the front end to CuraEngine? I see in the console that it passes the object/mesh indexes with the `l` flag, though it looks to me like the mesh has actually already been assembled with uranium as a `UM.Mesh.MeshData`?  That data already has a filename variable associated with it if I am looking at this correctly. Would it be possible to simply add that filename to the mesh object in CuraEngine so it could replace the memory address? Again, sorry if I am missing something here.

alekseisasin, 2018-07-16:
There is a difference between running CuraEngine in console and Cura. In Cura we are creating a message and then passing it to the CuraEngine. Check libArcus and libSavitar. 
One of the disadvantages of sending just a name, what if we have a few models with the same name, then we still need to distinguish them.
It is not difficult to add a clear name into G-code instead of a memory address. At this moment I cannot promise when it will be added/adjusted, but I will keep in mind your proposal.

paukstelis, 2018-07-26:
> One of the disadvantages of sending just a name, what if we have a few models with the same name, then we still need to distinguish them.
This was handled in Slic3r by just including `copy` followed by copy/index number. I think just appending the index number of the object should be sufficient.
I do think moving forward that including this information will be quite beneficial for real-time control of the printing processing.

BagelOrb, 2018-07-26:
It would be safer to add the mesh index rather than the memory address. I'm not a real hacker tho

paukstelis, 2018-08-21:
Pull request made. https://github.com/Ultimaker/CuraEngine/pull/828
As a non-programmer, I can say I have already learned a lot getting this far, though nearly all of these changes are just converting the memory address to a string.

Ghostkeeper, 2018-10-24:
I don't know why Github doesn't show https://github.com/Ultimaker/CuraEngine/pull/828 as being merged, but it is merged. This issue should be fixed for Cura 3.6.

closed 2018-10-24 by Ghostkeeper