Ultimaker/CuraEngine #815: Prime Tower introduces nozzle switches

BagelOrb, 2018-07-30:
**Application Version**
Cura master

**Platform**
Linux Mint - likely any

**Steps to Reproduce**
- load an object
- set it to the right extruder
- set the platform adhesion to the right extruder
- enable prime tower
see [second_extruder.curaproject.3mf.removeextension.zip](https://github.com/Ultimaker/CuraEngine/files/2241288/second_extruder.curaproject.3mf.removeextension.zip)

**Actual Results**
Nozzle switches on each layer!!!
[image](https://user-images.githubusercontent.com/8895761/43400311-b1bb393c-940d-11e8-9e59-dcf21cbcc1e1.png)

**Expected results**
No nozzle switches at all

**Additional Information**
Be careful that you never introduce nozzle switches without printing a prime tower when the prime tower is enabled.

Be careful not to print the prime tower and not to switch nozzle after the last layer which uses multiple materials, i.e. the nozzle shouldn't be switched any more if all next layers are with the same extruder.

Be careful that if you would only print the prime tower with a given extruder on a layer then that printing shouldn't count as a reason that we need a nozzle switch.

Ghostkeeper, 2018-08-01:
I can reproduce this problem with your project file. I also notice that it outputs a bunch of these to the log:

    2018-08-01 09:43:13,295 - DEBUG - [Thread-18] UM.Backend.Backend._backendLog [94]: [Backend] [WARNING] Unnecessary extruder switch detected! SliceDataStorage::getExtrudersUsed should probably be updated.

But when I try to reproduce the problem myself it doesn't occur: [DualExtrusionAdhesion.curaproject.3mf.removeextension.zip](https://github.com/Ultimaker/CuraEngine/files/2248477/DualExtrusionAdhesion.curaproject.3mf.removeextension.zip)

Or at least not initially. It did occur when I switched to TPU like BagelOrb's project file.

Something in the TPU profile is triggering this bug. I tried to look at a couple of the differences of TPU vs. PLA (e.g. Wipe Inactive Nozzle On Prime Tower) but couldn't reproduce it yet in the PLA profile. Maybe it's a combination of some settings that's causing this?

BagelOrb, 2018-08-01:
I would just look at the code and/or do a debug run with breakpoints,
rather than looking at which setting combinations trigger it.

This might very well be related to 204da21f5d9d2a710f29a667ba0f847f702749b2

BagelOrb, 2018-08-01:
[image]
This happens when you set the object to the second extruder in your project file.

This is specifically an issue when only the second extruder is used after some layer.

BagelOrb, 2018-08-01:
Ah no, it's definitely b3a0645c751e1a8589ae8611b53ae5a51d53d4bb

BagelOrb, 2018-08-01:
The comment `//The outermost prime tower extruder is always used if there is a prime tower.` is a dead giveaway ;)

diegopradogesto, 2018-08-02:
Yes, also system testing found this. It's in our sticky board. Very easy to reproduce by selecting TPU material in UM3 or UMS5.

BagelOrb, 2018-08-02:
TPU is irrelevant. It's just the combination of having Prime Tower enabled
and setting the model to the second extruder.

diegopradogesto, 2018-08-02:
Yes, I know, but it was the way to trigger the bug. I also thought that was related with the prime tower and testing the version before merging the feature removes the problem. But you were quicker and found the problem.

BagelOrb, 2018-08-02:
Haha ok. :+1:

Ghostkeeper, 2018-08-02:
Indeed that "Mark outermost prime tower extruder as used" commit is definitely the problem. Thanks for tracing that down. We must only mark it as used in layers lower than the height of the prime tower.

commit ee18d4ea (2018-08-02): Don't print prime tower higher than highest extruder switch

Fixes #815 and CURA-5615 and ST-1985.
closed 2018-08-02 by Ghostkeeper