Ultimaker/CuraEngine #834: [REQUEST] better bridging calculations for top surfaces

tmbdev, 2018-08-21:
Right now, it appears that Cura always follows blindly whatever pattern was specified for the top surface when putting down layers. For example, when printing a small hollow cube (no infill) with a concentric pattern for the top surface, the g-code will start extruding in mid-air. That's despite the fact that it would be easy in that case to generate g-code that bridges the top of the cube correctly.

It seems to me that Cura should never start extruding in mid-air, since that's just not going to work. It should either omit such extrusions from output (in which case people can see the resulting defects directly in layer view), or it should modify layers so that the bridging works.

I think the reason this isn't more of a problem right now is that zigzag layers tend to have this problem less. But they are not a good solution, because often the zigzag pattern is not ideal for bridging, and sometimes no layer pattern fits at all.

smartavionics, 2018-08-21:
Hi, have you tried out the experimental bridging settings yet? They force skins over air to use lines rather than concentric.

tmbdev, 2018-08-27:
Thanks, I hadn't see that, I'll give it a try.

However, as I indicated, I think no fixed pattern is optimal. Line patterns can end up mid-air as well even when they could bridge correctly at another angle.

Another common problem is a wide cylinder going to a narrow cylinder, which can be bridged pretty well when recognized by the slicer but doesn't work with lines.

Ghostkeeper, 2018-10-26:
I wouldn't know how we could ever design a strategy to choose the optimal pattern to bridge with. The only way I can think of is to actually generate every pattern and then measure the highest bridged length, and try to minimise that brute force, but that will surely be too computationally expensive.

bjude, 2018-10-29:
The only efficient way to bridge is a straight line, anything curved or with direction changes will have obvious issues.

Instead of generating every pattern and checking the highest bridge length, you could sample all the holes at some interval across them at different angles (say every 45 degrees) and then bisection search for the best angle between the 2 best. The infill pattern for that layer would then be set to 'lines', rotated to that angle.

For the case where there are a number of holes of different shapes, that layer could be split into sections that each have their infill calculated separately, as above. Splitting this way seems similar to a Voronoi diagram i think. Though this method would require all the extra machinery for generating several different infills in a given layer, possibly delimited by a pseudo 'inner wall' to cleanup any mismatch between the patterns along the joining line.