Ultimaker/CuraEngine #839: [master] Request - how to disable Stb update step at start of build

smartavionics, 2018-09-04:
**Application Version**

**Platform**

**Qt**

**PyQt**

**Display Driver**

**Steps to Reproduce**

**Actual Results**

**Expected results**

**Additional Information**

Ignoring template as it isn't relevant.

Everytime I build CuraEngine it does a git pull on the Stb repo which is all very well but it means that you can't actually build CuraEngine without an internet connection. So is it possible to build without doing that ?

smartavionics, 2018-09-05:
That's great, thanks for the swift fix. Closing...

closed 2018-09-05 by smartavionics

LipuFei, 2018-09-05:
@smartavionics Wait... I need one more fix, and you need to change `/usr/local/include/stb` to /usr/local/include/Stb` "S" in capital letter.

smartavionics, 2018-09-05:
why do we need to use Stb rather than stb?

LipuFei, 2018-09-05:
There: 4dade474

LipuFei, 2018-09-05:
Because CMake likes to checkout the repo using `Stb` unfortunately.

smartavionics, 2018-09-05:
OK, understood.

smartavionics, 2018-09-05:
Except this doesn't work right on Linux.

With /usr/local/include/stb, cmake can find the header files:

````
~/3dp/cura/CuraEngine.git/build[mb-testing]$ sh run-cmake.sh 
-- Building with Arcus
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found Protobuf: /opt/cura/lib/libprotobuf.a (Required is at least version "3.0.0") 
-- Found Stb installation at: /usr/local/include/stb
-- Compile with more optimization flags
-- Try OpenMP C flag = [-fopenmp]
-- Performing Test OpenMP_FLAG_DETECTED
-- Performing Test OpenMP_FLAG_DETECTED - Success
-- Try OpenMP CXX flag = [-fopenmp]
-- Performing Test OpenMP_FLAG_DETECTED
-- Performing Test OpenMP_FLAG_DETECTED - Success
-- Found OpenMP: -fopenmp  
-- Configuring done
CMake Warning (dev) at CMakeLists.txt:201 (add_dependencies):
  Policy CMP0046 is not set: Error on non-existent dependency in
  add_dependencies.  Run "cmake --help-policy CMP0046" for policy details.
  Use the cmake_policy command to set the policy and suppress this warning.

  The dependency target "Stb" of target "_CuraEngine" does not exist.
This warning is for project developers.  Use -Wno-dev to suppress it.
````

But using /usr/local/include/Stb fails to find the header files and it reverts back to using the local repo as before.

So now I have in /usr/local/include the stb repo and a symbolic link Stb that points at stb.

The stb detection in cmake wants the lower case version.

LipuFei, 2018-09-05:
Could you pull `master` again? Could be missing this commit: 4dade47

LipuFei, 2018-09-05:
Ah, I see. So find package Stb looks for `stb` instead...

smartavionics, 2018-09-05:
Thanks, works OK without the Stb symlink.