Ultimaker/CuraEngine #842: Build error with a recent version of CMake

thopiekar, 2018-09-08:
Hey guys,

I noticed failing builds on Ubuntu Cosmic, which is currently under development.
Builds fail across all architectures, so I looked into the logs and noticed that something goes wrong when copying a file.
However, the issue does not appear on a previous Ubuntu release, which uses an older version of CMake. Additionally, this does not appear in any other of your projects.
Therefore I guess there is something not compliant.

```
Get:64 http://ftpmaster.internal/ubuntu cosmic/main amd64 cmake amd64 3.12.1-1 [3418 kB]
```

```
[ 98%] Built target StringTest
[ 99%] Linking CXX executable CuraEngine
/usr/bin/cmake -E cmake_link_script CMakeFiles/CuraEngine.dir/link.txt --verbose=1
/usr/bin/c++   -O3 -DNDEBUG -Wall -static-libstdc++ -fopenmp  -Wl,-Bsymbolic-functions -Wl,-z,relro -rdynamic CMakeFiles/CuraEngine.dir/src/main.cpp.o  -o CuraEngine -Wl,-rpath,:::::::: lib_CuraEngine.a libclipper.a /usr/lib/x86_64-linux-gnu/libArcus.so.1.1.0 -lprotobuf -lpthread -lpthread 
/usr/bin/cmake -E copy_directory /<<PKGBUILDDIR>>/resources /<<PKGBUILDDIR>>/obj-x86_64-linux-gnu
Error copying directory from "/<<PKGBUILDDIR>>/resources" to "/<<PKGBUILDDIR>>/obj-x86_64-linux-gnu".
make[3]: *** [CMakeFiles/CuraEngine.dir/build.make:92: CuraEngine] Error 1
make[3]: *** Deleting file 'CuraEngine'
make[3]: Leaving directory '/<<PKGBUILDDIR>>/obj-x86_64-linux-gnu'
make[2]: *** [CMakeFiles/Makefile2:115: CMakeFiles/CuraEngine.dir/all] Error 2
make[2]: *** Waiting for unfinished jobs....
```

alekseisasin, 2018-09-10:
I had recently another issue after updating Ubuntu from 16.04 to 18.04.
I used to set: CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -no-pie" . But this is a special case
I know recently we made changes in _cmake_ related to STB library, @LipuFei can it be related to this change?

thopiekar, 2018-09-11:
Well, this issue should be not related to stb, because I have no internet access while building packages at launchpad.net. Therefore I created a separate package for stb.

Now I noticed that the issue only appears on 3.4. Looked into the CMakeLists.txt and noticed that there is a section, which tries to do the copy and probably whenever a directory is not present, then previous CMake versions were ignoring the command.
Couldn't find the commit, which removes these lines easily, so I'm going to prepare a PR for it. I never had any complaints from people using 3.4 with the 3.3 engine, but I think it would be great to provide the correct engine after getting this fixed in 3.4.

Ghostkeeper, 2018-09-12:
This was fixed in master here: https://github.com/Ultimaker/CuraEngine/commit/5aad55bf67e52ce5bdb27a3925af8a4cab441b38

closed 2018-09-12 by Ghostkeeper