Ultimaker/CuraEngine #847: [master] - optimising paths produces weird thin lines between infill

smartavionics, 2018-09-12:
**Application Version**

master

**Platform**

Linux

**Qt**

**PyQt**

**Display Driver**

**Steps to Reproduce**

**Actual Results**

**Expected results**

**Additional Information**

Just noticed these thin extrusions occurring between some infill lines when I would have expected travel moves instead.

With LayerPlan::optimizePaths(const Point& starting_position) being called you get...

[image]

Comment the call to that out and you get...

[image]

smartavionics, 2018-09-13:
[image]

rburema, 2018-09-13:
Hi @smartavionics, Which type of infill are you using? It might help (also to decide whether to work on it) if we could reproduce it with one of the 'standard' infill types. If this is only reproducible with non-'standard' types, chances are we might skip on this one. cheers, Remco.

smartavionics, 2018-09-13:
Hi @rburema , I'm using an experimental gyroid infill. I'll close this issue now and re-open it when the gyroid infill gets incorporated into Cura.

closed 2018-09-13 by smartavionics

Ghostkeeper, 2018-11-13:
Does this still occur with Gyroid infill?

smartavionics, 2018-11-13:
It's still badly broken but it looks to be in a different way as shown below. Those gyroid waves should not be connected and also notice how the relatively smooth waves turn into thinner, straighter lines.

BTW, for my own Cura, I have completely disabled the called to optimizePaths() as it breaks so much stuff.

[image]

Ghostkeeper, 2018-11-13:
The thinner, straighter lines is probably the MergeInfillLines class doing badly. We've also seen this and have a bug in our sprint for it.