Ultimaker/CuraEngine #848: Build on windows with mingw

lamest, 2018-09-13:
can someone please guide me throu this error?

**Application Version**
one of fresh commits with id=bbdeb43513c103aca8a6ec715d0ada38d89c0be6
and tagged comit with tag=3.4.1

**Platform**
docker image from microsoft/dotnet-framework:4.7.1-sdk

**Qt**
none

**PyQt**
latest

**Display Driver**
none

**Steps to Reproduce**
build with mingw from sources with protobuf and arcus

**Actual Results**
compilation failed with:
`CMakeFiles\_CuraEngine.dir\build.make:60: *** target pattern contains no '%'.  Stop.`
full log [here](https://gist.github.com/lamest/62388d431496bfbb0cc764c8b5c297ce)

the 60th str is: 
`Cura.pb.cc: ../protobuf::protoc`

protobuf is builded from sources with mingw. tested versions are 3.0.0 and 3.6.1

rburema, 2018-09-13:
Hi @lamest,

Looks like you're using almost the same command line as we're using (I'm pretty sure we'ven't got an 'make install' command for CuraEngine though, the normal procedure on Windows is to copy the resulting `.exe` to the Cura dir).

Assuming you've got everything else set-up correctly, the only thing that's a little bit different is the cmake command, which does have some extra parameters:
`cmake -DCMAKE_BUILD_TYPE=Release -DArcus_DIR=C:\dev\libArcus-mingw\install_dir\lib\cmake\Arcus -DPROTOBUF_SRC_ROOT_FOLDER=%PROTOBUF_MINGW_SRC% -DPROTOBUF_LIBRARY=%PROTOBUF_MINGW_INSTALL%/lib/libprotobuf.a -DPROTOBUF_INCLUDE_DIR=%PROTOBUF_MINGW_INSTALL%/include -DPROTOBUF_PROTOC_EXECUTABLE=%PROTOBUF_MINGW_INSTALL%/bin/protoc.exe -G "MinGW Makefiles" ..`
(Adapt to your own use of course...)

There is one other thing that might help (that springs to mind now): Use the 'VS2015 Native Command Tools Prompt' instead of the normal terminal. (Not sure how you'd do that in an automated docker package though.)

Hope that helps.
cheers, Remco.

lamest, 2018-09-20:
allright, looks like that was "\" in filepath that causes the error. just replaced them with "/"
but now i'm stuck with 
`cc1plus.exe: out of memory allocating 67112959 bytes`

[here](https://gist.github.com/lamest/26e81e8c1632f1f996a6e3f8778cd398)

rburema, 2018-09-20:
Haven't encountered that specific problem, and (since I don't think this issue looks like it's related to our code _specifically_) a quick google points to a few common issues. Are you using 32-bit Windows and/or a 32-bit compiler? If so, [this](https://github.com/monero-project/monero/issues/3521#issuecomment-377466568) comment I found elsewhere might help.

lamest, 2018-09-20:
no, both host machine and docker image are windows 10 x64. mingw is x86_64-8.1.0-release-posix-sjlj-rt_v6-rev0 package from  [sourceforge](https://sourceforge.net/projects/mingw-w64/files/)

what is your mingw version?

lamest, 2018-09-20:
allright, with mingw x86_64-7.3.0-release-posix-sjlj-rt_v5-rev0 its somehow magically works. I assume that this is pure random luck because sometimes i was missing like 60k of memory co compile with mingw 8.1.

rburema, 2018-09-20:
I'm using `8.1.0` as well, though not that particular one.

For future reference:
`PS C:\dev\git\CuraEngine> gcc --version`
`gcc.exe (i686-posix-dwarf-rev0, Built by MinGW-W64 project) 8.1.0`

Since this seems to work now, I'm closing this issue. If you have any objections, we can perhaps open it again.

closed 2018-09-20 by rburema

lamest, 2018-09-28:
for reference
looks like root of the problem was in docker. by default it should use all available memory for his containers, but it somehow failing randomly to do so.
solution is to explicitly set memory limit by `--memory=4g` and now its compiling with wingw 8.1

rburema, 2018-09-28:
@lamest, Nice, thanks for the callback. Too few people do that sort of thing. We should keep this in mind if we start using docker someday.