Ultimaker/CuraEngine #851: Skirt not going around all the objects on the build plate

dotps1, 2018-09-15:
**Application Version**

3.4.1

**Platform**

Windows 10 Pro x64

**Qt**

**PyQt**

**Display Driver**

18.30.11.01-180827a-332601E-RadeonSoftwareAdrenalin

**Steps to Reproduce**

I added 8 objects to the print bed, then used the "Arrange All Models" option.  

**Actual Results**

The skirt is going right through one of the objects.

[image]

**Expected results**

The skirt should be going around all the objects.

**Additional Information**

is there a way to manually adjust the  skirt placement?
[CCR10_X_Y_Tensioners.zip](https://github.com/Ultimaker/CuraEngine/files/2385795/CCR10_X_Y_Tensioners.zip)

dotps1, 2018-09-15:
I just wanted to add, if I move the object down -.5mm the skirt goes correctly, but it looks like its flat based on this:  
[image]
And if I manually move it with the mouse, it snaps down to the plate so I would think its flat and on the plate.

nallath, 2018-09-17:
The red just shows it's overhang, not that it's actually flat. The fact that it snaps on the build plate does not mean that the object is flat. It could well be that it used the lowest part to put it on the build plate. Based on the behavior that moving it a bit below it does work, my guess is that the bottom of the model isn't quite flat.

nallath, 2018-09-17:
You could check out the PrintOrientation plugin. Although it's not quite what you need, it will probably fix your issue. The plugin tries to find an orientation in such a way that it has as little overhang as possible. So if your print is already close to that, it will most likely also make the minute change so that it's as flat as possible on the buildplate.

rburema, 2018-09-21:
(In Cura-frontend:) There is 'Lay Flat' under the the 'Rotate' icon in the sidebar on the left, but that only works with the nearest face and when the model _can_ actually be laid flat. (Which means that it won't really work all too well when a single model is really a group of objects which all may or may not have some small visually almost invisible height-offset, because we of course shouldn't just 'break' a users' model.)

Since this isn't really a bug in the engine on examination, I'm going to close it. Feel free to make a feature request in Cura (frontend), but keep in mind that we could choose to defer it, or skip it completely, depending on all kinds of different factors, like available dev.-time.

closed 2018-09-21 by rburema