Ultimaker/CuraEngine #860: [Build error] Compile fails with latest CE Source.

KangDroid, 2018-09-27:
Ok, here is the one. 

I have Clang(Apple's default gcc) of "Apple LLVM version 10.0.0 (clang-1000.10.44.2)", and macOS Mojave(Command line tools)

And it is likely to fail build. Basically referencing to errors and commit log from GitHub, https://github.com/Ultimaker/CuraEngine/commit/fb8e44fe459d5a28846e42778160cadcf4ab63b9#diff-6965d87fb1b21ec7b221cb84f8389c12
this commit might be the one. 

Error log:
`[ 11%] Building CXX object CMakeFiles/_CuraEngine.dir/src/infill/SubDivCube.cpp.o
In file included from /Users/kangdroid/Desktop/temp/cura-build/build/CuraEngine-prefix/src/CuraEngine/src/infill/SubDivCube.cpp:4:
In file included from /Users/kangdroid/Desktop/temp/cura-build/build/CuraEngine-prefix/src/CuraEngine/src/infill/SubDivCube.h:7:
In file included from /Users/kangdroid/Desktop/temp/cura-build/build/CuraEngine-prefix/src/CuraEngine/src/infill/../sliceDataStorage.h:10:
In file included from /Users/kangdroid/Desktop/temp/cura-build/build/CuraEngine-prefix/src/CuraEngine/src/infill/../PrimeTower.h:12:
In file included from /Users/kangdroid/Desktop/temp/cura-build/build/CuraEngine-prefix/src/CuraEngine/src/infill/../utils/polygonUtils.h:11:
/Users/kangdroid/Desktop/temp/cura-build/build/CuraEngine-prefix/src/CuraEngine/src/infill/../utils/optional.h:171:20: warning: 'constexpr' non-static member function will not be implicitly 'const' in C++14; add 'const' to avoid a change in behavior [-Wconstexpr-not-const]
    constexpr bool operator==(const optional<U>& rhs)
                   ^
                                                      const
/Users/kangdroid/Desktop/temp/cura-build/build/CuraEngine-prefix/src/CuraEngine/src/infill/../utils/optional.h:133:9: warning: use of this statement in a constexpr function is a C++14 extension [-Wc++14-extensions]
        assert(instance && "Instance should be instantiated!");
        ^
/usr/include/assert.h:56:19: note: expanded from macro 'assert'
#define assert(e)       ((void)0)
                        ^
In file included from /Users/kangdroid/Desktop/temp/cura-build/build/CuraEngine-prefix/src/CuraEngine/src/infill/SubDivCube.cpp:4:
In file included from /Users/kangdroid/Desktop/temp/cura-build/build/CuraEngine-prefix/src/CuraEngine/src/infill/SubDivCube.h:7:
In file included from /Users/kangdroid/Desktop/temp/cura-build/build/CuraEngine-prefix/src/CuraEngine/src/infill/../sliceDataStorage.h:10:
In file included from /Users/kangdroid/Desktop/temp/cura-build/build/CuraEngine-prefix/src/CuraEngine/src/infill/../PrimeTower.h:12:
In file included from /Users/kangdroid/Desktop/temp/cura-build/build/CuraEngine-prefix/src/CuraEngine/src/infill/../utils/polygonUtils.h:11:
/Users/kangdroid/Desktop/temp/cura-build/build/CuraEngine-prefix/src/CuraEngine/src/infill/../utils/optional.h:138:9: warning: use of this statement in a constexpr function is a C++14 extension [-Wc++14-extensions]
        assert(instance && "Instance should be instantiated!");
        ^
/usr/include/assert.h:56:19: note: expanded from macro 'assert'
#define assert(e)       ((void)0)
                        ^
/Users/kangdroid/Desktop/temp/cura-build/build/CuraEngine-prefix/src/CuraEngine/src/infill/SubDivCube.cpp:255:22: error: constexpr variable 'sqrt_three_fourths' must be initialized by a constant expression
    constexpr double sqrt_three_fourths = sqrt(3.0 / 4.0);
                     ^                    ~~~~~~~~~~~~~~~
/Users/kangdroid/Desktop/temp/cura-build/build/CuraEngine-prefix/src/CuraEngine/src/infill/SubDivCube.cpp:255:43: note: non-constexpr function 'sqrt' cannot be used in a constant expression
    constexpr double sqrt_three_fourths = sqrt(3.0 / 4.0);
                                          ^
/usr/include/math.h:447:15: note: declared here
extern double sqrt(double);
              ^
3 warnings and 1 error generated.
make[5]: *** [CMakeFiles/_CuraEngine.dir/src/infill/SubDivCube.cpp.o] Error 1
make[4]: *** [CMakeFiles/_CuraEngine.dir/all] Error 2
make[3]: *** [all] Error 2
make[2]: *** [CuraEngine-prefix/src/CuraEngine-stamp/CuraEngine-build] Error 2
make[1]: *** [CMakeFiles/CuraEngine.dir/all] Error 2
make: *** [all] Error 2`

rburema, 2018-10-17:
Hi @KangDroid, This looks like it has been fixed in 77a6c8deee47bdb9476e48df2f8f331792640f06. cheers, Remco.

rburema, 2018-10-17:
I'm going to close this, if it isn't fixed by that commit, feel free to reopen it, or create a different issue.

closed 2018-10-17 by rburema