Ultimaker/CuraEngine #862: [3.4.1] Cura should look for the nearest island, but it does a travel

neukyhm, 2018-09-30:
**Application Version**
3.4.1

**Platform**
W10

**Steps to Reproduce**
Added a box for a machine to slice using the profile linked below.
[https://drive.google.com/open?id=12bRVfYncbCGy51SuVDdg34yMPs3bgex0](url)

**Actual Results**
Cura does a travel across the box (blue line in the picture linked below)
[https://drive.google.com/open?id=1FsQ8Hbxpjt_kysGVuk_ZDPsV5HdJyASv](url)

**Expected results**
Cura should do the red path I have drawn. When Cura finishes slicing an island, it should look for the next nearest island, but it won't continue to the nearest island, doing an unnecessary travel movement.

**Additional Information**
I think this is a big problem of Cura, because it wasted time traveling to an island that isn't the nearest.

alekseisasin, 2018-10-02:
@neukyhm , there might be not efficient travel moves and not optimal printing order. Can you attach your model to this issue and we will take a look for this specific case.

neukyhm, 2018-10-02:
> 
> 
> @neukyhm , there might be not efficient travel moves and not optimal printing order. Can you attach your model to this issue and we will take a look for this specific case.

Here you have: https://drive.google.com/open?id=1wZolCnl04tXAm2ESBnWtFvX-SWu3yB9Q
You will see that sometimes Cura does long travels when it could go to the nearest island with a shorter travel.

alekseisasin, 2018-10-02:
ok, I just tried Custom FDM printer, I saw the order not optimal. So, you try to change Layer Start X, Layer Start Y value. In some cases, it might help.

neukyhm, 2018-10-02:
> 
> 
> ok, I just tried Custom FDM printer, I saw the order not optimal. So, you try to change Layer Start X, Layer Start Y value. In some cases, it might help.

Tweaking those parameters helps a little, however I expected that Cura choose better the next island.

alekseisasin, 2018-10-03:
I agree it is a bit tricky, it was just a simple way how to get improvement.
The issue "order of printed areas is not optimal" is known.