Ultimaker/CuraEngine #873: CuraEngine 3.4.1, 3.5 flat models: 0 print time, 0 filament used

Viat0r, 2018-10-12:
**Description**
When slicing flat models (test.stl attached) I get 0 print time and 0 filament used (test.out.txt). However if I rotate the same model by 90 degrees (test_rotated.stl) to make it taller I get reasonable results (test_rotated.out.txt).
In GUI Cura everything's fine, although even if I copy all the -s parameters from cura.log I still get the same problem.

**Application Version**
3.4.1
3.5

**Platform**
Linux

**Steps to Reproduce**
Take test.stl from the attached zip file

CuraEngine slice -v -j '/usr/share/Cura/resources/definitions/fdmprinter.def.json' -v -o /dev/null -l test.stl

**Actual Results**

Cura 15.04: (good!)
Print time: 14206 
Filament: 13965

CuraEngine 3.4.1: (error)
Print time (s): 0
Print time (hr|min|s): 0h 0m 0s
Filament (mm^3): 0

CuraEngine 3.5: (error)
[ERROR] Unknown exception.

**Expected results**
Print Time > 0
Filament > 0

**Attachment**
[bug report.zip](https://github.com/Ultimaker/CuraEngine/files/2473214/bug.report.zip)

alekseisasin, 2018-11-08:
DEV: CURA-5835

Viat0r, 2018-11-08:
Thank you!
I tried different repair tools like slic3r, admesh, freecad and microsoft repair service, but they didn't seem to find any errors.
So I had to write my own tool for levelling the z coordinate.
This is weird, but I can live with it.