Ultimaker/CuraEngine #875: Layer Height control for outer perimeters

MrAmericanMike, 2018-10-12:
So I was thinking the other day, it would be really interesting to have an option where we can control the layer height for the outer perimeter / perimeters independently of the layer height of the overall print.

For example, been able to select a layer height for the print of 0,24 but been able to have control for the outer perimeter where I can specify "Layer Height / 2" or "Layer Height / 3"

This way we can make the print look nicer, on the outside, while still printing faster the infills and support layers.

I was thinking if this would even work... and I don't see a reason on why it shouldn't, lets say I have selected a print with 3 walls and then selected that I want all 3 of them (Not just the outer one, even when this could be an option too) to be Layer Height / 3, so in a 0,24 print I would have the 3 walls print first at 0,08 each, and then while at that 3 pass is done, do the 0,24 infills and supports and everything else.

Hope I make myself clear on the idea. And hopefully it's something that you can look into. Maybe I'm not original, maybe this exists on other slicer I really don't know, but I think it's a good idea and something to look into.

Thanks for your time. Let me know any question/feedback you have about this idea.

bjude, 2018-10-14:
I've thought about this before and i think theres an isssue with it for sloping surfaces but not sure how bad it really is. Either a gap will be left between the shell layers and the infill or there will be 'steps' that are the height of the inner layer thickness. This crappy photoshop will help explain (i realise the blue and red paths should be the same width but different height but this was a rush job):

[image]

On the left is generating the external paths (blue) at double the resolution of the inner paths(red). This causes a gap between the bottom shell layer and the inner paths. This will give a good print finish as the 'steps' are smaller but could cause issues with shells adhering to the infill.
On the right is the other method where the thicker layer height is used but its split in half which will leave steps as if we had just used the red path thickness for everything.

Given that infill overlaps shell layers somewhat i think the option on the left is probably viable but it would need some testing to be sure

DDDirk, 2018-10-15:
One solution would be to use overextrusion to make the shell-roads for those layers wider (and shift the roads a bit), but safer is the other way around: make the roads narrower and add a road. Thus:
[image]

DDDirk, 2018-10-18:
Ah, yes, the images are misleading because most of the time the infill will be at an angle to the shell. It _does_ have to have the right amount of overlap with it, though, and the sze of the gap changes with the overhang-angle and layerheight - in my drawings 45° and 0.1 mm for the shell (if the trace width is 0.4 mm). But with higher layer heights a doubled layer height for the infill would be a bad idea anyway.

But there is another issue. Changes in extrusion speed (can) have a negative effect on strength (I am currently investigating that). This appears to even be the case with small changes. In your proposal, the extrusion speed would double when going from shell to infill and halve the other way around. That is assuming an equal print speed. But infill is often printed faster, which in itself would be a problem, and this would make it even worse.

However, the fact that is may have issues is not enough reason not to implement it (else you could never try anything new). Maybe print speed and/or looks matter more to someone than strength. Also, it would offer a way to test how big these issues are.
So it will depend very much on how much work would be required to implement this, and that I can not judge.

bjude, 2018-10-20:
So this might already be a feature, see "Infill Layer Thickness": https://ultimaker.com/en/resources/52670-infill

MrAmericanMike, 2018-10-25:
> 
> 
> So this might already be a feature, see "Infill Layer Thickness": https://ultimaker.com/en/resources/52670-infill

:O Ohhh wow!!! Didn't know it was already a thing. I will have to try this out now.
Thanks a lot for the heads up.

I guess we can close the "Issue" now, great to know it was already a thing.

Ghostkeeper, 2018-12-19:
Different layer thickness for outer vs. inner wall has been considered but it's a very complex problem because walls might not align properly above each other. It would be nice to use a lower layer height for only the outer walls but you're then limited by the horizontal resolution of your line thickness. It's also going to give problems to change the flow suddenly to twice as much while still printing something that's visible on the outside.

Let's hope that the infill layer thickness and support layer thickness we have now are sufficient.

closed 2018-12-19 by Ghostkeeper