Ultimaker/CuraEngine #885: Unwanted temperature changes on a dual extruder with one hotend

PaulF8080, 2018-10-21:
My new creatity CR-X came with an old cura version and I am trying to add it to 3.51. This happens on "other" printers, too. I don't want the trapezoid temperature profile used for two hot ends.  It bloats the gcode and the expansion and contraction hurts the heater life expectancy. I wrote a perl script to count the M commands for my model.
M82     3
M84     1
M104    1257
M105    419
M106    1
M107    3
M109    419
M117    1
M140    1

I thought maybe I missed a setting so I cloned the engine git. I found code  in the dual extruder train "for" loop at line 1347 of LayerPlan.cpp that I don't understand:

```
            { // require printing temperature to be met
                constexpr bool wait = true;
                gcode.writeTemperatureCommand(extruder_nr, extruder_plan.required_start_temperature, wait);
            }

```
A brace pair with no conditional. Could it please have an "if" added that checks a new parameter or that the nozzle offsets are the same?

PaulF8080, 2018-10-22:
I know about scope. Why would someone define a variable only used once and only used in the function on the next line. If the person wants show the last parameter is true, assigning a variable is very strange, 
 way to me. I try not to criticize other people's coding style.  it compiles into instructions  and data unless the compiler optimizes it out.

bjude, 2018-10-23:
Its done in a few places in the codebase, I find it a bit confusing too. Whoever wrote it seems to want to pass named variables to functions instead of calling `foo(true, false, false, true)`. Its somewhat clearer to have named variables and call `foo(wait, print_result, nuke_whales, save_earth)`. Not the way i'd do it but it's not particularly uncommon in this project

PaulF8080, 2018-10-23:
The thing is that if it is known the compiler optimizes it out, it's brilliant.

smartavionics, 2018-10-23:
I think the reason why it is done (and I have done it myself) is that it's much easier to come back to the code later and understand a call to a function that looks like this:
````
const double accuracy = 3.6;
const long understanding = 20;
const bool readability = true;
do_something_nice(accuracy, understanding, readability);
````
Compared to this:
````
do_something_nice(3.6, 20, true);
````

PaulF8080, 2018-10-23:
That's exactly why I like it so much.. It speaks much clearer than a long comment.  I learned something.

PaulF8080, 2018-10-23:
Back to the issue. I believe the reason heater filaments fail is because the heat changes flex the metal causing metal fatigue failures. II guess some people think static high temperature  is  the  problem.  Hence the need for a new setting to please people with both opinions.  Ask yourself why light bulbs mostly fail when you are switching them on.