Ultimaker/CuraEngine #894: Prime tower does not prime second extruder before printing

notnyt, 2018-10-28:
**Application Version**
3.5.1

**Platform**
win 64

**Steps to Reproduce**
setup a print with two extruders using a prime tower

**Actual Results**
prime tower only primes first extruder on first layer

**Expected results**
prime tower should prime second nozzle before use, but it does not

**Additional Information**
You can see the prime tower on the left only primed the first extruder, and not the second.

[image]

rburema, 2018-11-08:
Hi @notnyt,
Could you attach the model for which this happens?
cheers, Remco.

notnyt, 2018-11-09:
[test.zip](https://github.com/Ultimaker/CuraEngine/files/2564560/test.zip)
attached

Ghostkeeper, 2019-01-22:
I've sliced your 3MF file using default settings of the UM3, but enabling the prime tower and support and setting the support extruder to the second extruder. This produced a file with two G280 commands, one at the beginning and the other just after the first switch to T1. So that seems to work fine.

Maybe "Enable Prime Blob" was disabled for the second extruder?

notnyt, 2019-01-23:
> I've sliced your 3MF file using default settings of the UM3, but enabling the prime tower and support and setting the support extruder to the second extruder. This produced a file with two G280 [...]

It's not an option that even shows up when you add a custom printer.

[image]

Ghostkeeper, 2019-01-23:
Ah, sorry. I'm so used to working with UM3 and UMS5 that it slipped my mind.

Reading through your original issue report though I find that your expected result doesn't match mine. The prime tower is meant to prime an extruder after being on stand-by. The first layer should have a skirt or a brim to prime for you, but that is not visible in your screenshot. Maybe you disabled platform adhesion?

notnyt, 2019-01-23:
Of course I can't remember now since this was months ago, but I think there was some trouble using skirts to prime at that time.  Looking at how it slices now, it looks fine, though.

closed 2019-02-19 by Ghostkeeper