Ultimaker/CuraEngine #902: New Feature request: Open Bottom at build surface (Without affecting any other downskins)

benwynne, 2018-11-05:
Hello All,

I have what I believe is a new feature request. 

**Would it be possible to specify an open bottom for a print?**

I can kind of achieve what I want by specifying a "Bottom Thickness" of 0. However this affects all down-skins, not just the one intersecting the build platform.

I'm guessing this'd be an easy add to skin.cpp. Can anyone point me in the right direction?

-B

**Application Version**

3.5.1

**Platform**

All

**Qt**

**PyQt**

**Display Driver**

**Steps to Reproduce**

**Actual Results**

**Expected results**

**Additional Information**

rburema, 2018-11-08:
Hi @benwynne,
Yes, feature request is the correct term :-)

Personally, for something like this, I'd adapt the model, or remove the bottom later, rather than trying to force the engine to do it this way. That said; perhaps you could also set 'initial layer height' to a very small value. _However_, be aware that this could have consequences for adhesion to the buildplate.
cheers, Remco.

ChrisTerBeke, 2018-11-08:
You could try setting the amount of bottom layers to 0 or the bottom layers infill percentage to match your normal infill percentage? I've seen people do this and the result matches your described target result.

Ghostkeeper, 2019-03-08:
This would be implemented by https://github.com/Ultimaker/CuraEngine/pull/872.