Ultimaker/CuraEngine #905: CuraEngine Infill Parameter

gavogar, 2018-11-13:
Hi Team, how are?
I have a question abourt the correct param to use Cura with Infill. 

CuraEngine slice -v -j [...] -s **infill="xx"** where xx is in %.

What is the correct param? Can i get a list with the params?

Thanks!

alekseisasin, 2018-11-13:
Hi @gavogar ,
The easiest way to find all params is: 
Every time when you slice in Cura GUI, it sends all params to CuraEngine and stores in a log file.

gavogar, 2018-11-13:
Hi @alekseisasin,
I can understand that you told me. But I want a dynamic form because I am using CuraEngine in a server and send differents params to re-evaluate the printing time.
Thanks!

alekseisasin, 2018-11-14:
Dynamic HTML form, you want to have a form with setting fields. After changing a value in a field you want to directly send a group of affected settings to CuraEngine. You should keep in mind many settings influence other settings. If you change one setting it will affect other. All these things happen before sending the message to CuraEngine.

To understand the process of sending the message (data) to CuraEngine you can check:
https://github.com/Ultimaker/Cura/blob/master/plugins/CuraEngineBackend/CuraEngineBackend.py
https://github.com/Ultimaker/Cura/blob/master/plugins/CuraEngineBackend/StartSliceJob.py

alekseisasin, 2018-11-16:
These setting which showed on the screen is: "infill_sparse_density". I think this is the answer to your question.
All these settings you can find in this file:
https://github.com/Ultimaker/Cura/blob/master/resources/definitions/fdmprinter.def.json

But changing one param might not give you accurate data, because in GUI the related params are updated automatically. All settings relations you can find as well in the same document
https://github.com/Ultimaker/Cura/blob/master/resources/definitions/fdmprinter.def.json
(Note, all these settings are inherited by printer profile and can be overwritten)

closed 2024-04-05 by wawanbreton