Ultimaker/CuraEngine #914: zigzag infill does not always function properly

notnyt, 2018-11-15:
**Application Version**
3.5.1

**Platform**
windows

**Steps to Reproduce**
select zig zag for top and bottom skin.  5 star gear bearing pictured

**Actual Results**

[image]

**Expected results**
The skin should cover the entire layer

smartavionics, 2018-11-15:
Hello @notnyt , I think that is a regression introduced in 3.5. As far as I am aware, it's still not been addressed even in 3.6 (may be wrong about that). My Cura builds have the naughty code disabled so they should work OK. If you want to try, use the latest release at https://www.dropbox.com/sh/s43vqzmi4d2bqe2/AAADdYdSu9iwcKa0Knqgurm4a?dl=0

rburema, 2018-11-21:
Hi all, This is related to the 'fat infill' (casued by spurious merging) fixes, we'll try to keep this in mind as well as we're working on that. (Devs: see CURA-5776) cheers, Remco.

alekseisasin, 2018-11-22:
I close the ticket because it is duplicated issue: https://github.com/Ultimaker/Cura/issues/4524

closed 2018-11-22 by alekseisasin

notnyt, 2018-11-22:
> 
> 
> I close the ticket because it is duplicated issue: [Ultimaker/Cura#4524](https://github.com/Ultimaker/Cura/issues/4524)

That's ironing, though, this is infill

rburema, 2018-11-23:
It may not be clear from the descriptions, but the underlying issue is the same in all of these cases: too agressive merging.