Ultimaker/CuraEngine #926: Error while building Cura Engine (Windows 10) - problem with protobuf

barnej121, 2018-12-05:
Hi everyone,

I have a problem with build Cura Engine. There is an error while building a Cura engine :

Scanning dependencies of target CuraEngine                                                                                                                                                                                                   [ 97%] Building CXX object CMakeFiles/CuraEngine.dir/src/main.cpp.obj                                                                                                                                                                        [ 98%] Building RC object CMakeFiles/CuraEngine.dir/CuraEngine.rc.obj                                                                                                                                                                        [100%] Linking CXX executable CuraEngine.exe                                                                                                                                                                                                 "C:\Program Files\CMake\bin\cmake.exe" -E remove -f CMakeFiles\CuraEngine.dir/objects.a                                                                                                                                                      C:\PROGRA~1\MINGW\mingw64\bin\ar.exe cr CMakeFiles\CuraEngine.dir/objects.a @CMakeFiles\CuraEngine.dir\objects1.rsp                                                                                                                          C:\PROGRA~1\MINGW\mingw64\bin\G__~1.EXE  -O3 -DNDEBUG -Wall -fopenmp  -m64 -Wl,--whole-archive CMakeFiles\CuraEngine.dir/objects.a -Wl,--no-whole-archive  -o CuraEngine.exe -Wl,--out-implib,libCuraEngine.dll.a -Wl,--major-image-version,0,--minor-image-version,0 @CMakeFiles\CuraEngine.dir\linklibs.rsp                                                                                                                                                                             Warning: corrupt .drectve at end of def file                                                                                                                                                                                                 Warning: corrupt .drectve at end of def file                                                                                                                                                                                                 Warning: corrupt .drectve at end of def file                                                                                                                                                                                                 Warning: corrupt .drectve at end of def file                                                                                                                                                                                                 Warning: corrupt .drectve at end of def file                                                                                                                                                                                                 lib_CuraEngine.a(ArcusCommunication.cpp.obj):ArcusCommunication.cpp:(.text+0x117): undefined reference to `google::protobuf::Arena::AllocateAligned(std::type_info const*, unsigned long long)'                                              lib_CuraEngine.a(ArcusCommunication.cpp.obj):ArcusCommunication.cpp:(.text+0x14a): undefined reference to `google::protobuf::Arena::AddListNode(void*, void (*)(void*))'                                                                     lib_CuraEngine.a(ArcusCommunication.cpp.obj):ArcusCommunication.cpp:(.text+0x247): undefined reference to `google::protobuf::Arena::AllocateAligned(std::type_info const*, unsigned long long)'                                              lib_CuraEngine.a(ArcusCommunication.cpp.obj):ArcusCommunication.cpp:(.text+0x27a): undefined reference to `google::protobuf::Arena::AddListNode(void*, void (*)(void*))'                                                                     lib_CuraEngine.a(ArcusCommunication.cpp.obj):ArcusCommunication.cpp:(.text+0xb54): undefined reference to `Arcus::Socket::Socket()'                                                                                                          lib_CuraEngine.a(ArcusCommunication.cpp.obj):ArcusCommunication.cpp:(.text+0xb9d): undefined reference to `Arcus::Socket::addListener(Arcus::SocketListener*)'                                                                               lib_CuraEngine.a(ArcusCommunication.cpp.obj):ArcusCommunication.cpp:(.text+0x19cd): undefined reference to `google::protobuf::internal::RepeatedPtrFieldBase::Reserve(int)'                                                                  lib_CuraEngine.a(ArcusCommunication.cpp.obj):ArcusCommunication.cpp:(.text+0x2b3e): undefined reference to `google::protobuf::internal::RepeatedPtrFieldBase::Reserve(int)'                                                                  lib_CuraEngine.a(ArcusCommunication.cpp.obj):ArcusCommunication.cpp:(.text+0x37ed): undefined reference to `google::protobuf::internal::RepeatedPtrFieldBase::Reserve(int)'                                                                  lib_CuraEngine.a(ArcusCommunication.cpp.obj):ArcusCommunication.cpp:(.text+0x3fad): undefined reference to `google::protobuf::internal::RepeatedPtrFieldBase::Reserve(int)'                                                                  lib_CuraEngine.a(ArcusCommunication.cpp.obj):ArcusCommunication.cpp:(.text+0x471d): undefined reference to `google::protobuf::internal::RepeatedPtrFieldBase::Reserve(int)'                                                                  lib_CuraEngine.a(ArcusCommunication.cpp.obj):ArcusCommunication.cpp:(.rdata$.refptr._ZN6google8protobuf8internal13empty_string_B5cxx11E[.refptr._ZN6google8protobuf8internal13empty_string_B5cxx11E]+0x0): undefined reference to `google::protobuf::internal::empty_string_[abi:cxx11]'                                                                                                                                                                                                  lib_CuraEngine.a(Listener.cpp.obj):Listener.cpp:(.text+0x8c): undefined reference to `Arcus::Error::getErrorCode() const'                                                                                                                    lib_CuraEngine.a(Listener.cpp.obj):Listener.cpp:(.text+0xb9): undefined reference to `Arcus::Error::getErrorMessage[abi:cxx11]() const'                                                                                                      lib_CuraEngine.a(Listener.cpp.obj):Listener.cpp:(.text+0x124): undefined reference to `Arcus::Error::getErrorMessage[abi:cxx11]() const'                                                                                                     lib_CuraEngine.a(Cura.pb.cc.obj):Cura.pb.cc:(.text+0x47c): undefined reference to `google::protobuf::internal::WireFormatLite::SkipField(google::protobuf::io::CodedInputStream*, unsigned int)'                                             lib_CuraEngine.a(Cura.pb.cc.obj):Cura.pb.cc:(.text+0x498): undefined reference to `google::protobuf::io::CodedInputStream::ReadTagFallback(unsigned int)'                                                                                    lib_CuraEngine.a(Cura.pb.cc.obj):Cura.pb.cc:(.text+0x4b6): undefined reference to `google::protobuf::io::CodedInputStream::ReadTagFallback(unsigned int)'                                                                                    lib_CuraEngine.a(Cura.pb.cc.obj):Cura.pb.cc:(.text+0x508): undefined reference to `google::protobuf::internal::WireFormatLite::WriteMessageMaybeToArray(int, google::protobuf::MessageLite const&, google::protobuf::io::CodedOutputStream*)'lib_CuraEngine.a(Cura.pb.cc.obj):Cura.pb.cc:(.text+0x538): undefined reference to `google::protobuf::internal::WireFormatLite::WriteMessageMaybeToArray(int, google::protobuf::MessageLite const&, google::protobuf::io::CodedOutputStream*)'lib_CuraEngine.a(Cura.pb.cc.obj):Cura.pb.cc:(.text+0x588): undefined reference to `google::protobuf::internal::WireFormatLite::WriteMessageMaybeToArray(int, google::protobuf::MessageLite const&, google::protobuf::io::CodedOutputStream*)'lib_CuraEngine.a(Cura.pb.cc.obj):Cura.pb.cc:(.text+0x5d8): undefined reference to `google::protobuf: [... cut ...] eHandler3'                                                                                                                                                                  C:/cura-build-environment/build/inst/lib/libprotobuf.lib(CMakeFiles/libprotobuf.dir/C_/cura-build-environment/build/Protobuf-prefix/src/Protobuf/src/google/protobuf/stubs/int128.cc.obj):(.xdata[$unwind$?catch$0@?0??_Osfx@?$basic_ostream@DU?$char_traits@D@std@@@std@@QEAAXXZ@4HA]+0x8): undefined reference to `__CxxFrameHandler3'

Ghostkeeper, 2018-12-05:
You are linking CuraEngine to a version of Protobuf that was not built with MinGW, I suspect.

barnej121, 2018-12-05:
Hmm I don't know, but I have all libraries in the Protobuf-MinGW catalog libprotobuf.a libprotoc.a and libprotobuf-lite.a. I built protobuf separately for MSVC and MinGW but I still get the same error. Maybe a bad path or something is invisible ..

LipuFei, 2019-09-24:
The linking is not correct. From your log, you can see that you are compiling and linking CuraEngine with MinGW, but it's linking against `libprotobuf.lib`, which is the protobuf compiled with MSVC. You need to make sure that your configuration is correct.