Ultimaker/CuraEngine #930: [3.6.0] Slicing for second extruder does not work like before.

ipepe, 2018-12-12:
**Application Version**
3.6.0

**Platform**
Mac OS X - Darwin

**Qt**
Official Ultimaker website

**PyQt**
Official Ultimaker website

**Display Driver**
Intel Iris Pro 1536 MB graphics

**Steps to Reproduce**
`CuraEngine slice -j dual_extruder_printer.def.json -s adhesion_extruder_nr=1 -e1 -l 'file.stl' -o 'output.gcode'`

**Actual Results**
output.gcode is printing on `T0` extruder

**Expected results**
output.gcode is printing on `T1` extruder

**Additional Information**
I can create repository with example `dual_extruder_printer.def.json` if needed. I suppose there were changes regarding `extruders_enabled_count` and probably some more values that I should consider in my def.json?

ipepe, 2019-01-03:
Well that sounds great... But that also means that I should compile Windows, Linux and Mac version of library. And that does not sound so great...

If You are aware of this issue, then why it's not merged?

rburema, 2019-01-15:
_Possible_ duplicate of (frontend) [issue](https://github.com/Ultimaker/Cura/issues/5103), in which case it was fixed already (see #974, will be in the _second_ 4.0-BETA, which isn't out yet).

@ipepe, Please note that @WinstonMao isn't part of the team and can't merge anything into our repo here.

ipepe, 2019-01-15:
Thanks for information.

ipepe, 2019-05-13:
Today I tried slicing on second extruder with supports and adhesion on T1 and I did not succeed.

rburema, 2019-05-14:
Hi @ipepe, Could you send `dual_extruder_printer.def.json` and the input .stl? You can post them here if you rename the files to have a .zip extension, so you don't have to make an entire repository just for a bug-report.

ipepe, 2019-05-14:
Hi @rburema. 
I created https://github.com/ipepe/CuraEngine-second-extruder-slicing for You to look into.

rburema, 2019-05-14:
Hi @ipepe Thanks for the effort. I can see that in the `machine_start_gcode` value hardcodes values for `T`, the tool-switch command. We give a `T` command _before_ the start-gcode, so that the start-gcode can run on the selected extruder.

In general, at this moment, we don't really support hard-coding extruder-switch commands, though someting like this can be realized with `T{initial_extruder_nr}\n`. You can't select the 'off' extruder this way though, since there might be more than 2 extruders.

See also [this](https://github.com/Ultimaker/Cura/issues/5502), a related issue.

ipepe, 2019-05-14:
@rburema Thanks for explanation

closed 2019-05-15 by rburema

ipepe, 2019-05-15:
@rburema Wierd thing is that when sliced using same start/end gcode using Full Cura (Desktop GUI) it manages to set T1 just after starting Layer -6