Ultimaker/CuraEngine #938: [3.6.0] Gradual Infill generates print on thin air

soswow, 2018-12-28:
**Application Version**
3.6.0

**Platform**
Mac

**Steps to Reproduce**
Using supplied stl file use 25% density, Grid pattern and Gradual infill (as shown on screenshot). Layer height is (0.15)

**Actual Results**
Supporting for the hole Infill for 33 layers starting from 77 are printed on thin air.

**Expected results**
At list *some* lines printed starting from the bottom to have something to base evergrowing density of infill above.

[image]
[image]
[image]

[Body1.stl.zip](https://github.com/Ultimaker/CuraEngine/files/2714730/Body1.stl.zip)

soswow, 2018-12-28:
In the example above it *could* work out since ever-increasing density of top most level will allow hole's bottom skin to catch up. But I make it taller, I will end up with completely useless case.
[Body2.stl.zip](https://github.com/Ultimaker/CuraEngine/files/2714740/Body2.stl.zip)
[image]
[image]
[image]

Ghostkeeper, 2019-01-16:
This is expected (but not desired) behaviour right now. The problem is the same as when you would fill in 1.5625% infill and gradual infill steps to 0: You just have so low an infill density that a lot of stuff falls in between the infill lines.

I'd suggest setting the gradual infill steps to 3 instead of 4, so that down below the infill density is a mere 1/8th of the normal infill density of 25%, rather than 1/16th.