Ultimaker/CuraEngine #971: Compilation with VS2017

chaplin89, 2019-01-03:
Hi,
I just tried to compile CuraEngine with VS but it fails on [this line](https://github.com/Ultimaker/CuraEngine/commit/5b67ca1a00e840c6e8cd8900c9f5a1a4d629f628#diff-928fbc1ff2e7746c2692bd243ff28bd5R47).

It should not be standard C++ code AFAIK, can you tell me what compiler are you using?
Also, I saw this pattern very often in your codebase, almost always followed by std::fill_n() or a memset().

Out of curiosity, leaving apart the conformance to the c++ standard, why just don't use an std::vector<> using the second overload of this list http://www.cplusplus.com/reference/vector/vector/vector ?
It should be more clear/concise and it behave almost identically to an initialization of a POD array.
Also, it will avoid performance penalties related to the resizing of the vector.

bjude, 2019-01-07:
The project is compiled with MinGW under windows and doesnt support MSVC (currently) due to more than just that line..

I have a PR open (#823) that fixes this but its out of date now. Once i get a chance i'll update it and that should fix the Windows compilation issues. Based on the comment in that PR it doesnt look like Ultimaker want to add MSVC to their test suite so it may rebreak, which is a shame.

bjude, 2019-03-05:
This should now be fixed since #823 is merged, try building the master branch

Ghostkeeper, 2019-03-05:
Indeed we've confirmed that this works with VS2017 now.

We (=Ultimaker) don't have any interest to keep up support for alternative compilers. That's extra work for us to maintain that's just of no real value to us. Bjude just worked through 8 years of backlog regarding maintaining that so hopefully if it breaks again it will be easier to fix if people really want to use MSVC to compile CuraEngine.

closed 2019-03-05 by Ghostkeeper

bjude, 2019-03-06:
While I understand that there are no current plans to add MSVC support (officially), it may be worth thinking about in the future since libArcus already requires being built with MSVC (to link with python) and MinGW (to link with CuraEngine). If CuraEngine was built with MSVC on windows the double compile would no longer be necessary.

Obviously switching compilers and build processes isnt trivial but there could be real benefits to having MSVC as a first-class citizen

Ghostkeeper, 2019-03-06:
Yes, but that would then mean that we'd drop support for MinGW, and thus require a closed-source compiler.

sedwards2009, 2019-03-06:
...and also for any of our Python modules which contain native code.

We're already crossed the closed-source compiler bridge.

Ghostkeeper, 2019-03-06:
For Cura's front-end, yes, purely by virtue of using CPython. CuraEngine doesn't depend on CPython though.