Ultimaker/CuraEngine #978: [3.6.0] Wall line count ignored for thin walls

mihai-dumitresq, 2019-01-20:
**Application Version**
3.6.0

**Steps to Reproduce**
Set Wall Count to 2 then prepare. I'm using https://www.thingiverse.com/thing:2975429 to test but scaled to 50% of the size.

**Actual Results**
Thin walls have the outer wall but no inner wall, even though it would fit. And instead of an inner wall, a skin is printed. I've tried multiple options and I can't get the inner wall to print there (even if it would be just 1 inner wall instead of 2), like "Compensate Wall Overlaps", "Outer Before Inner Wall", "Print Thin Walls":
[image]

**Expected results**
Thin walls have the outer wall and some inner wall, if it fits.

**Additional Information**
Also note how jittery the travel/print path for that thin skin on the right wall. It's creating a lot of vibration when trying to print said part on an Ender 3, even with travel&print jerk as low as 8mm/s.
I've tried "Extra skin wall count" option to try to get the outer skin to be printed similarly to how the inner wall would be but that doesn't work as per issue #977.

smartavionics, 2019-01-20:
Yes, Ultimaker's Cura doesn't make a good job of either wall gap filling or thin walls. My experimental releases have a new implementation which generally performs better. You can find them here https://www.dropbox.com/sh/s43vqzmi4d2bqe2/AAADdYdSu9iwcKa0Knqgurm4a?dl=0.

[image]

mihai-dumitresq, 2019-01-23:
@smartavionics: I've tried it out and thin walls print great with your experimental release (Cura-mb-master-20190118.AppImage). Thank you!
Is there a plan to port those changes in into Ultimaker's Cura?

smartavionics, 2019-01-23:
>  Is there a plan to port those changes in into Ultimaker's Cura?

Quite probably I will submit them at some point in the future. However, I already have around 10 contributions (PRs) pending that are currently not receiving any attention from the Cura developers so I am in no rush to add another. Perhaps if they clear the backlog of PRs, I will do it.