Ultimaker/CuraEngine #979: CLI breaking changes from 3.6 to 4.0?

ipepe, 2019-01-21:
Hello.

Will CuraEngine in version 4.0 will be much different from 3.x regarding Command Line Interface?

I built custom wrapper around CuraEngine 3.x and wanted to know if I should schedule major rewrites of my code to comply with version 4.0?

rburema, 2019-02-27:
Most of the focus for 4.0 was on the new user workflow, new interface, Marketplace, cloud, etc. in Cura itself. The engine/slicer side has been relatively quiet and mostly recieved a few community PR's and some bug-fixes. I can't know how your wrapper works, but it seems a safe bet to say only minimal changes will be requied, if any.

closed 2019-02-27 by rburema

ipepe, 2019-02-27:
@rburema Could You estimate when stable 4.0 will be officially released?

Ghostkeeper, 2019-02-27:
Our plan is to release the stable version in about 3 weeks.

Ghostkeeper, 2019-02-27:
You're going to need to add defaults for the settings that have been added since then. I don't know an exhaustive list, but there's at least these three, to give an example: https://github.com/Ultimaker/Cura/commit/de8e56ae19647b58208aec4b305be7aafffd868a#diff-be0cfced2c55852f4bce442bd971db52

If you run CuraEngine it'll give an error quickly enough, if you've missed a setting. Or you could pluck fdmprinter.def.json from the Cura repository and pass that along, depending on how your wrapper works.

ipepe, 2019-02-27:
@Ghostkeeper Thanks. I will review changes between fdmprinter.def.json currently used in our wrapper and 4.0. Thank You for all information.