Ultimaker/CuraEngine #980: [3.6/4.0] Spurious movements to origin, multiextrusion only?

paukstelis, 2019-01-25:
I'm doing multiextrusion printing and I have found that on certain models there are movements to the origin (X0,Y0) that I cannot explain from any settings that I have adjusted. It usually happens at beginning of prints and after the prime pillar, though when I've sliced without a prime pillar I also see them. This happens on both 3.6 and 4.0. Here is an example (M616 is a custom gcode command for this printer):

```
G0 F3000 X0.00 Y0.00 Z0.6  <-----HERE
;LAYER:1
G1 F1200 E-3
M82 ;absolute extrusion mode
M104 S180
G91; relative
G0 Z1;
G90; absolute
G0 X300 Y144 F8000
M83 ;relative extrusion mode
T0
M82 ;absolute extrusion mode
G91; relative
G0 X-5 Y-10
M616 T0
M109 S195
G90; absolute
M83 ;relative extrusion mode
M105
M109 S195
M104 T2 S180
M106 S255
;MESH:untitled-tiamat-backplate.stl
G0 F6000 X0.00 Y0.00 Z0.8   <-------HERE
G0 X0.00 Y0.00 Z1.1              <-------HERE
G1 F1200 E3
G0 F6000 X0.00 Y10 Z0.8      <-------This is weird too
G1 F300 Z0.8
```
I have merged models to get them aligned, and then ungrouped to assign the appropriate extruder. Not sure if that has something to do with this. Any insight would be welcome.

paukstelis, 2019-01-26:
Looking a little further, I noticed that there are other unexplained movements even in gcode that doesn't move to the origin. In this example, there is a G0 movement over the part prior to going to the prime tower (which starts with ;SUPPORT). 

> G1 F1200 E3
> G1 F300 Z8.6
> ;MESH:Object 7
> **G0 F6000 X274.279 Y143.434 Z8.6**
> **G0 X267.72 Y144.72**
> M104 S205
> ;TYPE:SUPPORT
> G1 F300 Z8.3
> G1 F3000 E3.02652
> G1 F2400 X299.28 Y144.72 E1.25963
> G1 X299.28 Y176.28 E1.25963
> G1 X267.72 Y176.28 E1.25963
> G1 X267.72 Y145.386 E1.23305
> G0 F2160 X267.72 Y144.72
> G1 F3000 E-3
> G1 F300 Z8.6
> G0 F6000 X265.964 Y220.165 Z8.6
> ;TYPE:WALL-INNER
> G1 F300 Z8.3
> G1 F3000 E3.02658
> G1 F2400 X266.006 Y220.258 E0.00339

paukstelis, 2019-01-31:
Here is another example with screenshot to visualize what is happening. There is no reason for T2 to move back over the already printed part before it prints the skirt. The tool heads has moved back to the prime tower position before the tool change takes place.

[image]
Associated gcode:

```
G0 F3000 X260.433 Y218.572
G1 F1200 X264.735 Y222.874 E0.30353
G0 F3000 X264.78 Y222.353
G1 F1200 X260.529 Y218.102 E0.29993
G0 F3000 X260.663 Y217.671
G1 F1200 X264.791 Y221.799 E0.29125
G0 F3000 X264.778 Y221.22
G1 F1200 X260.827 Y217.269 E0.27876
G0 F3000 X261.044 Y216.921
G1 F1200 X264.758 Y220.634 E0.26201
G0 F3000 X264.72 Y220.031
G1 F1200 X261.304 Y216.615 E0.24102
G0 F3000 X261.618 Y216.363
G1 F1200 X264.66 Y219.406 E0.21466
G0 F3000 X264.561 Y218.741
G1 F1200 X262.032 Y216.212 E0.17843
G0 F3000 X262.586 Y216.2
G1 F1200 X264.405 Y218.019 E0.12834
G1 X264.475 Y218.089
G1 F3000 E-3
G1 F300 Z0.6
G0 F3000 X263.912 Y216.96 Z0.6
G1 F300 Z0.3
G1 F3000 E3
G1 F1200 X263.446 Y216.494 E0.03288
G1 F1200 E-7
M82 ;absolute extrusion mode
M104 S180
G91; relative
G0 Z1;
G90; absolute
G0 X300 Y144 F8000
M83 ;relative extrusion mode
T2
M82 ;absolute extrusion mode
G91; relative
G0 X-15 Y-10
M616 T2
M109 S195
G90; absolute
M83 ;relative extrusion mode
M105
M109 S205
M104 T1 S195
;MESH:NONMESH
G0 F3000 X263.446 Y216.494 Z0.3
G1 F3000 E-3
G0 F3000 X263.446 Y226.494
G1 F300 Z0.6
G0 F3000 X255.251 Y227.307 Z0.6
```

closed 2019-03-09 by paukstelis

Ghostkeeper, 2019-03-11:
Duplicate of https://github.com/Ultimaker/Cura/issues/5383.