Ultimaker/CuraEngine #984: [3.6.0] GCC9 build failures: OpenMP data sharing

hroncok, 2019-01-29:
**Application Version**
3.6.0

**Platform**
Linux (Fedora 30) with GCC 9.0.1.

**Qt**
not relevant

**PyQt**
not relevant

**Display Driver**
not relevant

**Steps to Reproduce**
build CuraEngine 3.6.0 with gcc 9.

**Actual Results**

```
[ 78%] Building CXX object CMakeFiles/_CuraEngine.dir/src/utils/gettime.cpp.o
/usr/bin/c++  -DARCUS -DVERSION=\"3.5.1\" -I/builddir/build/BUILD/CuraEngine-3.5.1/./stb-e6afb9cbae4064da8c3e69af3ff5c4629579c1d2 -I/usr/include/polyclipping -I/builddir/build/BUILD/CuraEngine-3.5.1  -O2 -g -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -fexceptions -fstack-protector-strong -grecord-gcc-switches -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -fPIC -O2 -DNDEBUG -Wall -fopenmp   -std=gnu++11 -o CMakeFiles/_CuraEngine.dir/src/utils/gettime.cpp.o -c /builddir/build/BUILD/CuraEngine-3.5.1/src/utils/gettime.cpp
BUILDSTDERR: /builddir/build/BUILD/CuraEngine-3.5.1/src/layerPart.cpp: In function 'void cura::createLayerParts(cura::SliceMeshStorage&, cura::Slicer*, bool, bool)':
BUILDSTDERR: /builddir/build/BUILD/CuraEngine-3.5.1/src/layerPart.cpp:50:46: error: 'total_layers' not specified in enclosing 'parallel'
BUILDSTDERR:    50 |     for (unsigned int layer_nr = 0; layer_nr < total_layers; layer_nr++)
BUILDSTDERR:       |                                     ~~~~~~~~~^~~~~~~~~~~~~~
BUILDSTDERR: /builddir/build/BUILD/CuraEngine-3.5.1/src/layerPart.cpp:49:9: error: enclosing 'parallel'
BUILDSTDERR:    49 | #pragma omp parallel for default(none) shared(mesh,slicer) firstprivate(union_layers,union_all_remove_holes) schedule(dynamic)
BUILDSTDERR:       |         ^~~
[ 79%] Building CXX object CMakeFiles/_CuraEngine.dir/src/utils/LinearAlg2D.cpp.o
/usr/bin/c++  -DARCUS -DVERSION=\"3.5.1\" -I/builddir/build/BUILD/CuraEngine-3.5.1/./stb-e6afb9cbae4064da8c3e69af3ff5c4629579c1d2 -I/usr/include/polyclipping -I/builddir/build/BUILD/CuraEngine-3.5.1  -O2 -g -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -fexceptions -fstack-protector-strong -grecord-gcc-switches -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -fPIC -O2 -DNDEBUG -Wall -fopenmp   -std=gnu++11 -o CMakeFiles/_CuraEngine.dir/src/utils/LinearAlg2D.cpp.o -c /builddir/build/BUILD/CuraEngine-3.5.1/src/utils/LinearAlg2D.cpp
BUILDSTDERR: make[2]: *** [CMakeFiles/_CuraEngine.dir/build.make:217: CMakeFiles/_CuraEngine.dir/src/layerPart.cpp.o] Error 1
BUILDSTDERR: make[2]: *** Waiting for unfinished jobs....
BUILDSTDERR: /builddir/build/BUILD/CuraEngine-3.5.1/src/support.cpp: In static member function 'static void cura::AreaSupport::generateOverhangAreasForMesh(cura::SliceDataStorage&, cura::SliceMeshStorage&)':
BUILDSTDERR: /builddir/build/BUILD/CuraEngine-3.5.1/src/support.cpp:790:144: error: 'max_dist_from_lower_layer' not specified in enclosing 'parallel'
BUILDSTDERR:   790 |         std::pair<Polygons, Polygons> basic_and_full_overhang = computeBasicAndFullOverhang(storage, mesh, layer_idx, max_dist_from_lower_layer);
BUILDSTDERR:       |                                                                                                                                                ^
BUILDSTDERR: /builddir/build/BUILD/CuraEngine-3.5.1/src/support.cpp:787:13: error: enclosing 'parallel'
BUILDSTDERR:   787 |     #pragma omp parallel for default(none) shared(storage, mesh) schedule(dynamic)
BUILDSTDERR:       |             ^~~
BUILDSTDERR: /builddir/build/BUILD/CuraEngine-3.5.1/src/support.cpp: In static member function 'static void cura::AreaSupport::generateSupportAreasForMesh(cura::SliceDataStorage&, const cura::SettingsBaseVirtual&, const cura::SettingsBaseVirtual&, const cura::SettingsBaseVirtual&, size_t, size_t, std::vector<cura::Polygons>&)':
BUILDSTDERR: /builddir/build/BUILD/CuraEngine-3.5.1/src/support.cpp:843:48: error: 'layer_count' not specified in enclosing 'parallel'
BUILDSTDERR:   843 |     for (unsigned int layer_idx = 1; layer_idx < layer_count; layer_idx++)
BUILDSTDERR:       |                                      ~~~~~~~~~~^~~~~~~~~~~~~
BUILDSTDERR: /builddir/build/BUILD/CuraEngine-3.5.1/src/support.cpp:842:13: error: enclosing 'parallel'
BUILDSTDERR:   842 |     #pragma omp parallel for default(none) shared(xy_disallowed_per_layer, storage, mesh) schedule(dynamic)
BUILDSTDERR:       |             ^~~
BUILDSTDERR: /builddir/build/BUILD/CuraEngine-3.5.1/src/support.cpp:846:14: error: 'is_support_mesh_place_holder' not specified in enclosing 'parallel'
BUILDSTDERR:   846 |         if (!is_support_mesh_place_holder)
BUILDSTDERR:       |              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
BUILDSTDERR: /builddir/build/BUILD/CuraEngine-3.5.1/src/support.cpp:842:13: error: enclosing 'parallel'
BUILDSTDERR:   842 |     #pragma omp parallel for default(none) shared(xy_disallowed_per_layer, storage, mesh) schedule(dynamic)
BUILDSTDERR:       |             ^~~
BUILDSTDERR: /builddir/build/BUILD/CuraEngine-3.5.1/src/support.cpp:848:17: error: 'use_xy_distance_overhang' not specified in enclosing 'parallel'
BUILDSTDERR:   848 |             if (use_xy_distance_overhang) //Z overrides XY distance.
BUILDSTDERR:       |                 ^~~~~~~~~~~~~~~~~~~~~~~~
BUILDSTDERR: /builddir/build/BUILD/CuraEngine-3.5.1/src/support.cpp:842:13: error: enclosing 'parallel'
BUILDSTDERR:   842 |     #pragma omp parallel for default(none) shared(xy_disallowed_per_layer, storage, mesh) schedule(dynamic)
BUILDSTDERR:       |             ^~~
BUILDSTDERR: /builddir/build/BUILD/CuraEngine-3.5.1/src/support.cpp:851:104: error: 'z_distance_top' not specified in enclosing 'parallel'
BUILDSTDERR:   851 |                 Polygons xy_overhang_disallowed = mesh.overhang_areas[layer_idx].offset(z_distance_top * tan_angle);
BUILDSTDERR:       |                                                                                         ~~~~~~~~~~~~~~~^~~~~~~~~~~
BUILDSTDERR: /builddir/build/BUILD/CuraEngine-3.5.1/src/support.cpp:842:13: error: enclosing 'parallel'
BUILDSTDERR:   842 |     #pragma omp parallel for default(none) shared(xy_disallowed_per_layer, storage, mesh) schedule(dynamic)
BUILDSTDERR:       |             ^~~
BUILDSTDERR: /builddir/build/BUILD/CuraEngine-3.5.1/src/support.cpp:851:106: error: 'tan_angle' not specified in enclosing 'parallel'
BUILDSTDERR:   851 |                 Polygons xy_overhang_disallowed = mesh.overhang_areas[layer_idx].offset(z_distance_top * tan_angle);
BUILDSTDERR:       |                                                                                                          ^~~~~~~~~
BUILDSTDERR: /builddir/build/BUILD/CuraEngine-3.5.1/src/support.cpp:842:13: error: enclosing 'parallel'
[... 12 lines cut ...]
BUILDSTDERR:   842 |     #pragma omp parallel for default(none) shared(xy_disallowed_per_layer, storage, mesh) schedule(dynamic)
BUILDSTDERR:       |             ^~~
BUILDSTDERR: /builddir/build/BUILD/CuraEngine-3.5.1/src/support.cpp:1054:9: error: 'max_checking_idx_size_t' not specified in enclosing 'parallel'
BUILDSTDERR:  1054 |         for (size_t layer_idx = 0; layer_idx < max_checking_idx_size_t; layer_idx++)
BUILDSTDERR:       |         ^~~
BUILDSTDERR: /builddir/build/BUILD/CuraEngine-3.5.1/src/support.cpp:1053:9: error: enclosing 'parallel'
BUILDSTDERR:  1053 | #pragma omp parallel for default(none) shared(support_areas, storage) schedule(dynamic)
BUILDSTDERR:       |         ^~~
BUILDSTDERR: /builddir/build/BUILD/CuraEngine-3.5.1/src/support.cpp:1056:111: error: 'layer_z_distance_top' not specified in enclosing 'parallel'
BUILDSTDERR:  1056 |             support_areas[layer_idx] = support_areas[layer_idx].difference(storage.getLayerOutlines(layer_idx + layer_z_distance_top - 1, false));
BUILDSTDERR:       |                                                                                                     ~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~
BUILDSTDERR: /builddir/build/BUILD/CuraEngine-3.5.1/src/support.cpp:1053:9: error: enclosing 'parallel'
BUILDSTDERR:  1053 | #pragma omp parallel for default(none) shared(support_areas, storage) schedule(dynamic)
BUILDSTDERR:       |         ^~~
BUILDSTDERR: make[2]: *** [CMakeFiles/_CuraEngine.dir/build.make:438: CMakeFiles/_CuraEngine.dir/src/support.cpp.o] Error 1
```

Full build log including exact commands as they were run: [build.log](https://github.com/Ultimaker/CuraEngine/files/2809055/build.log.txt)

**Expected results**
CuraEngine should build.

**Additional Information**

https://gcc.gnu.org/gcc-9/porting_to.html#ompdatasharing

Ghostkeeper, 2019-02-01:
I would rather make the variable shared rather than private, so more like `shared(mesh, slicer, total_layers)` rather than `shared(mesh, slicer) firstprivate(total_layers)`. `firstprivate` creates a copy on every thread, which would then be faster to locally modified but since it is const anyway you won't need a copy of the variable on every thread.

The approach is correct though.

hroncok, 2019-02-01:
<shameless plug>I find Fedora very easy to use actually, you should give it a try some day.</shameless plug>.

I'll try to move everything I've added to firstprivate to shared and see if that builds.

hroncok, 2019-02-01:
```patch
diff --git a/src/layerPart.cpp b/src/layerPart.cpp
index 80370fe..07d40f2 100644
--- a/src/layerPart.cpp
+++ b/src/layerPart.cpp
@@ -49,7 +49,7 @@ void createLayerParts(SliceMeshStorage& mesh, Slicer* slicer)
 {
     const auto total_layers = slicer->layers.size();
     assert(mesh.layers.size() == total_layers);
-#pragma omp parallel for default(none) shared(mesh, slicer) schedule(dynamic)
+#pragma omp parallel for default(none) shared(mesh, slicer, total_layers) schedule(dynamic)
     for (unsigned int layer_nr = 0; layer_nr < total_layers; layer_nr++)
     {
         SliceLayer& layer_storage = mesh.layers[layer_nr];
diff --git a/src/support.cpp b/src/support.cpp
index a1081ad..1638c95 100644
--- a/src/support.cpp
+++ b/src/support.cpp
@@ -854,7 +854,7 @@ void AreaSupport::generateSupportAreasForMesh(SliceDataStorage& storage, const S
     const double tan_angle = tan(angle) - 0.01;  // the XY-component of the supportAngle
     xy_disallowed_per_layer[0] = storage.getLayerOutlines(0, false).offset(xy_distance);
     // for all other layers (of non support meshes) compute the overhang area and possibly use that when calculating the support disallowed area
-    #pragma omp parallel for default(none) shared(xy_disallowed_per_layer, storage, mesh) schedule(dynamic)
+    #pragma omp parallel for default(none) shared(xy_disallowed_per_layer, storage, mesh, layer_count, is_support_mesh_place_holder, use_xy_distance_overhang, z_distance_top, tan_angle, xy_distance, xy_distance_overhang) schedule(dynamic)
     for (unsigned int layer_idx = 1; layer_idx < layer_count; layer_idx++)
     {
         Polygons outlines = storage.getLayerOutlines(layer_idx, false);
@@ -1054,7 +1054,7 @@ void AreaSupport::generateSupportAreasForMesh(SliceDataStorage& storage, const S
         const int max_checking_layer_idx = std::min(static_cast<int>(storage.support.supportLayers.size())
                                                   , static_cast<int>(layer_count - (layer_z_distance_top - 1)));
         const size_t max_checking_idx_size_t = std::max(0, max_checking_layer_idx);
-#pragma omp parallel for default(none) shared(support_areas, storage) schedule(dynamic)
+#pragma omp parallel for default(none) shared(support_areas, storage, max_checking_idx_size_t, layer_z_distance_top) schedule(dynamic)
         for (size_t layer_idx = 0; layer_idx < max_checking_idx_size_t; layer_idx++)
         {
             support_areas[layer_idx] = support_areas[layer_idx].difference(storage.getLayerOutlines(layer_idx + layer_z_distance_top - 1, false));

```

This builds. AFAIK it should build on ancient Ubuntu gcc version as well, but could you please try?

Ghostkeeper, 2019-02-01:
Hmm, no, it doesn't seem to work on GCC 8.2.0. I get this error:

    [ 15%] Building CXX object CMakeFiles/_CuraEngine.dir/src/LayerPlan.cpp.o
    /home/ruben/Projects/CuraEngine/src/layerPart.cpp: In function ‘void cura::createLayerParts(cura::SliceMeshStorage&, cura::Slicer*)’:
    /home/ruben/Projects/CuraEngine/src/layerPart.cpp:53:92: error: ‘total_layers’ is predetermined ‘shared’ for ‘shared’
     #pragma omp parallel for default(none) shared(mesh, slicer, total_layers) schedule(dynamic)
                                                                                                ^

This is correct, because const variables should already be shared (though I would expect more of a warning rather than an error).

Ghostkeeper, 2019-02-04:
`firstprivate` compiles though. I'm not sure if that is a solution that we can live with because it'll reduce performance.

opoplawski, 2019-02-07:
I believe you could also drop default(none) and explicitly declare the shared const variables shared.

Ghostkeeper, 2019-02-07:
That still gives the same answer:

    [  9%] Building CXX object CMakeFiles/_CuraEngine.dir/src/layerPart.cpp.o
    /home/ruben/Projects/CuraEngine/src/layerPart.cpp: In function ‘void cura::createLayerParts(cura::SliceMeshStorage&, cura::Slicer*)’:
    /home/ruben/Projects/CuraEngine/src/layerPart.cpp:52:78: error: ‘total_layers’ is predetermined ‘shared’ for ‘shared’
     #pragma omp parallel for shared(mesh, slicer, total_layers) schedule(dynamic)
                                                                                  ^

opoplawski, 2019-02-13:
Ah, apparently I'm wrong about explicitly declaring it shared.  Leave it out of the shared() clause.  The default is to be shared so it's the same thing - just lacking some explicitness.

msva, 2019-06-05:
Is it any chance this will be fixed soon (at, least, there will be some patch for user-side applying)?
For now, it's impossible to build curaengine with all the modern versions of compiler:
GCC-9.1 says the above, and clang-8 says:
```make
/var/tmp/portage/media-gfx/curaengine-4.1.0/work/CuraEngine-4.1.0/src/utils/logoutput.cpp:30:9: error: 'va_start' cannot be used in a captured statement
        va_start(args, fmt);
        ^
/usr/lib64/llvm/8/bin/../../../../lib/clang/8.0.0/include/stdarg.h:33:29: note: expanded from macro 'va_start'
#define va_start(ap, param) __builtin_va_start(ap, param)
                            ^
/var/tmp/portage/media-gfx/curaengine-4.1.0/work/CuraEngine-4.1.0/src/utils/logoutput.cpp:43:9: error: 'va_start' cannot be used in a captured statement
        va_start(args, fmt);
        ^
/usr/lib64/llvm/8/bin/../../../../lib/clang/8.0.0/include/stdarg.h:33:29: note: expanded from macro 'va_start'
#define va_start(ap, param) __builtin_va_start(ap, param)
                            ^
/var/tmp/portage/media-gfx/curaengine-4.1.0/work/CuraEngine-4.1.0/src/utils/logoutput.cpp:56:9: error: 'va_start' cannot be used in a captured statement
        va_start(args, fmt);
        ^
/usr/lib64/llvm/8/bin/../../../../lib/clang/8.0.0/include/stdarg.h:33:29: note: expanded from macro 'va_start'
#define va_start(ap, param) __builtin_va_start(ap, param)
                            ^
/var/tmp/portage/media-gfx/curaengine-4.1.0/work/CuraEngine-4.1.0/src/utils/logoutput.cpp:71:9: error: 'va_start' cannot be used in a captured statement
        va_start(args, fmt);
        ^
/usr/lib64/llvm/8/bin/../../../../lib/clang/8.0.0/include/stdarg.h:33:29: note: expanded from macro 'va_start'
#define va_start(ap, param) __builtin_va_start(ap, param)
                            ^
/var/tmp/portage/media-gfx/curaengine-4.1.0/work/CuraEngine-4.1.0/src/utils/logoutput.cpp:87:9: error: 'va_start' cannot be used in a captured statement
        va_start(args, fmt);
        ^
/usr/lib64/llvm/8/bin/../../../../lib/clang/8.0.0/include/stdarg.h:33:29: note: expanded from macro 'va_start'
#define va_start(ap, param) __builtin_va_start(ap, param)
```

:crying_cat_face:

StefanBruens, 2019-06-16:
Please see:
https://www.gnu.org/software/gcc/gcc-9/porting_to.html - OpenMP data sharing

Ghostkeeper, 2019-06-17:
One of our developers (@LipuFei) just upgraded his MinGW to the version representing GCC 9.1.0. @LipuFei are you building with OpenMP and have you encountered this?

llukas, 2019-07-03:
@Ghostkeeper seems to fail on gentoo with gcc 9.1.0

gcc --version
gcc (Gentoo 9.1.0-r1 p1.1) 9.1.0

```
var/tmp/portage/media-gfx/curaengine-3.6.0/work/CuraEngine-3.6.0/src/layerPart.cpp: In function ‘void cura::createLayerParts(cura::SliceMeshStorage&, cura::Slicer*)’:
/var/tmp/portage/media-gfx/curaengine-3.6.0/work/CuraEngine-3.6.0/src/layerPart.cpp:53:46: error: ‘total_layers’ not specified in enclosing ‘parallel’
   53 |     for (unsigned int layer_nr = 0; layer_nr < total_layers; layer_nr++)
      |                                     ~~~~~~~~~^~~~~~~~~~~~~~
/var/tmp/portage/media-gfx/curaengine-3.6.0/work/CuraEngine-3.6.0/src/layerPart.cpp:52:9: error: enclosing ‘parallel’
   52 | #pragma omp parallel for default(none) shared(mesh, slicer) schedule(dynamic)
      |         ^~~
make[2]: *** [CMakeFiles/_CuraEngine.dir/build.make:214: CMakeFiles/_CuraEngine.dir/src/layerPart.cpp.o] Error 1
```

LipuFei, 2019-07-03:
I have the same issue with MinGW GCC 9.1.0. Making the changes suggested by the error messages fixes the issue, but those changes will break with earlier GCC versions.

LipuFei, 2019-07-03:
@StefanBruens Thanks for the info! Very useful!

sl1pkn07, 2019-07-08:
any rebased patch?

can't apply in recent commits

greetings

hroncok, 2019-07-08:
https://src.fedoraproject.org/rpms/CuraEngine/blob/master/f/CuraEngine-gcc9.patch has been rebased to 4.1.0

sl1pkn07, 2019-07-08:
is wreid, with my locales (es_ES.UTF-8)

~~~
[ 96%] Building CXX object CMakeFiles/_CuraEngine.dir/Cura.pb.cc.o

during GIMPLE pass: wrestrict
En la función ‘void InitDefaultsPolygon_Cura_2eproto()’:
Violación de segmento
  193 | static void InitDefaultsPolygon_Cura_2eproto() {
      |             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Por favor, envíe un informe completo de errores,
con el código preprocesado si es apropiado.
Véase <https://bugs.archlinux.org/> para instrucciones.
make[2]: *** [CMakeFiles/_CuraEngine.dir/build.make:1111: CMakeFiles/_CuraEngine.dir/Cura.pb.cc.o] Error 1
make[2]: *** Se espera a que terminen otras tareas....
make[1]: *** [CMakeFiles/Makefile2:74: 
~~~

with LC_ALL=C:

~~~
[ 96%] Building CXX object CMakeFiles/_CuraEngine.dir/Cura.pb.cc.o
In member function 'void cura::proto::Polygon::SharedCtor()',
    inlined from 'cura::proto::Polygon::Polygon()' at /tmp/makepkg/sl1-curaengine-git/src/build/Cura.pb.cc:3105:13,
    inlined from 'void InitDefaultsPolygon_Cura_2eproto()' at /tmp/makepkg/sl1-curaengine-git/src/build/Cura.pb.cc:3103:1:
/tmp/makepkg/sl1-curaengine-git/src/build/Cura.pb.cc:3126:11: warning: 'void* memset(void*, int, size_t)' offset [29, 40] from the object at 'cura::proto::_Polygon_default_instance_' is out of the bounds of referenced subobject 'cura::proto::Polygon::type_' with type 'int' at offset 24 [-Warray-bounds]
 3126 |   ::memset(&type_, 0, static_cast<size_t>(
      |   ~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
 3127 |       reinterpret_cast<char*>(&line_feedrate_) -
      |       ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
 3128 |       reinterpret_cast<char*>(&type_)) + sizeof(line_feedrate_));
      |       ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In member function 'void cura::proto::PathSegment::SharedCtor()',
    inlined from 'cura::proto::PathSegment::PathSegment()' at /tmp/makepkg/sl1-curaengine-git/src/build/Cura.pb.cc:3976:13,
    inlined from 'void InitDefaultsPathSegment_Cura_2eproto()' at /tmp/makepkg/sl1-curaengine-git/src/build/Cura.pb.cc:3974:1:
/tmp/makepkg/sl1-curaengine-git/src/build/Cura.pb.cc:4017:11: warning: 'void* memset(void*, int, size_t)' offset [61, 64] from the object at 'cura::proto::_PathSegment_default_instance_' is out of the bounds of referenced subobject 'cura::proto::PathSegment::extruder_' with type 'int' at offset 56 [-Warray-bounds]
 4017 |   ::memset(&extruder_, 0, static_cast<size_t>(
      |   ~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
 4018 |       reinterpret_cast<char*>(&point_type_) -
      |       ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
 4019 |       reinterpret_cast<char*>(&extruder_)) + sizeof(point_type_));
      |       ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In member function 'void cura::proto::Extruder::SharedCtor()',
    inlined from 'cura::proto::Extruder::Extruder()' at /tmp/makepkg/sl1-curaengine-git/src/build/Cura.pb.cc:1525:13,
    inlined from 'void InitDefaultsExtruder_Cura_2eproto()' at /tmp/makepkg/sl1-curaengine-git/src/build/Cura.pb.cc:1523:1:
/tmp/makepkg/sl1-curaengine-git/src/build/Cura.pb.cc:1544:11: warning: 'void* memset(void*, int, size_t)' offset [25, 28] from the object at 'cura::proto::_Extruder_default_instance_' is out of the bounds of referenced subobject 'cura::proto::Extruder::settings_' with type 'cura::proto::SettingList*' at offset 16 [-Warray-bounds]
 1544 |   ::memset(&settings_, 0, static_cast<size_t>(
      |   ~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
 1545 |       reinterpret_cast<char*>(&id_) -
      |       ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
 1546 |       reinterpret_cast<char*>(&settings_)) + sizeof(id_));
      |       ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In member function 'void cura::proto::PrintTimeMaterialEstimates::SharedCtor()',
    inlined from 'cura::proto::PrintTimeMaterialEstimates::PrintTimeMaterialEstimates()' at /tmp/makepkg/sl1-curaengine-git/src/build/Cura.pb.cc:4886:13,
    inlined from 'void InitDefaultsPrintTimeMaterialEstimates_Cura_2eproto()' at /tmp/makepkg/sl1-curaengine-git/src/build/Cura.pb.cc:4884:1:
/tmp/makepkg/sl1-curaengine-git/src/build/Cura.pb.cc:4903:11: warning: 'void* memset(void*, int, size_t)' offset [45, 88] from the object at 'cura::proto::_PrintTimeMaterialEstimates_default_instance_' is out of the bounds of referenced subobject 'cura::proto::PrintTimeMaterialEstimates::time_none_' with type 'float' at offset 40 [-Warray-bounds]
 4903 |   ::memset(&time_none_, 0, static_cast<size_t>(
      |   ~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
 4904 |       reinterpret_cast<char*>(&time_prime_tower_) -
      |       ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
 4905 |       reinterpret_cast<char*>(&time_none_)) + sizeof(time_prime_tower_));
      |       ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In member function 'void cura::proto::LayerOptimized::SharedCtor()',
    inlined from 'cura::proto::LayerOptimized::LayerOptimized()' at /tmp/makepkg/sl1-curaengine-git/src/build/Cura.pb.cc:3562:13,
    inlined from 'void InitDefaultsLayerOptimized_Cura_2eproto()' at /tmp/makepkg/sl1-curaengine-git/src/build/Cura.pb.cc:3560:1:
/tmp/makepkg/sl1-curaengine-git/src/build/Cura.pb.cc:3579:11: warning: 'void* memset(void*, int, size_t)' offset [45, 52] from the object at 'cura::proto::_LayerOptimized_default_instance_' is out of the bounds of referenced subobject 'cura::proto::LayerOptimized::id_' with type 'int' at offset 40 [-Warray-bounds]
[... cut ...]
 2711 |       reinterpret_cast<char*>(&thickness_) -
      |       ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
 2712 |       reinterpret_cast<char*>(&id_)) + sizeof(thickness_));
      |       ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[ 97%] Linking CXX static library lib_CuraEngine.a
[ 97%] Built target _CuraEngine
Scanning dependencies of target CuraEngine
[ 98%] Building CXX object CMakeFiles/CuraEngine.dir/src/main.cpp.o
[100%] Linking CXX executable CuraEngine
[100%] Built target CuraEngine
~~~

???

LipuFei, 2019-07-09:
From https://www.gnu.org/software/gcc/gcc-9/porting_to.html, looks like the only option to keep it compatible with both GCC <=8 and >=9 is to remove `default(none)`. I tested this with both GCC versions and it works.

onitake, 2019-08-15:
Debian has just raised the importance of this issue to "serious": https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=925664

@LipuFei Have you done further successful testing with `default(none)`?
In that case, I will add a patch to our ancient Cura version until we're ready to update to a newer release.

onitake, 2019-08-15:
This is the patch I want to apply (note that we're still on Cura 3.3, but not much has changed in the relevant code):

```patch
--- CuraEngine.orig/src/layerPart.cpp   2019-08-15 12:47:55.000000000 +0200
+++ CuraEngine/src/layerPart.cpp        2019-08-15 12:57:49.000000000 +0200
@@ -46,7 +46,7 @@
 {
     const auto total_layers = slicer->layers.size();
     assert(mesh.layers.size() == total_layers);
-#pragma omp parallel for default(none) shared(mesh,slicer) firstprivate(union_layers,union_all_remove_holes) schedule(dynamic)
+#pragma omp parallel for shared(mesh,slicer) firstprivate(union_layers,union_all_remove_holes) schedule(dynamic)
     for (unsigned int layer_nr = 0; layer_nr < total_layers; layer_nr++)
     {
         SliceLayer& layer_storage = mesh.layers[layer_nr];
--- CuraEngine.orig/src/support.cpp     2019-08-15 12:47:55.000000000 +0200
+++ CuraEngine/src/support.cpp  2019-08-15 12:58:42.590264508 +0200
@@ -720,7 +720,7 @@
     const double support_angle = mesh.getSettingInAngleRadians("support_angle");
     const double tan_angle = tan(support_angle) - 0.01;  //The X/Y component of the support angle. 0.01 to make 90 degrees work too.
     const coord_t max_dist_from_lower_layer = tan_angle * layer_height; //Maximum horizontal distance that can be bridged.
-    #pragma omp parallel for default(none) shared(storage, mesh) schedule(dynamic)
+    #pragma omp parallel for shared(storage, mesh) schedule(dynamic)
     for (unsigned int layer_idx = 1; layer_idx < storage.print_layer_count; layer_idx++)
     {
         std::pair<Polygons, Polygons> basic_and_full_overhang = computeBasicAndFullOverhang(storage, mesh, layer_idx, max_dist_from_lower_layer);
@@ -775,7 +775,7 @@
     const double tan_angle = tan(angle) - 0.01;  // the XY-component of the supportAngle
     xy_disallowed_per_layer[0] = storage.getLayerOutlines(0, false).offset(xy_distance);
     // for all other layers (of non support meshes) compute the overhang area and possibly use that when calculating the support disallowed area
-    #pragma omp parallel for default(none) shared(xy_disallowed_per_layer, storage, mesh) schedule(dynamic)
+    #pragma omp parallel for shared(xy_disallowed_per_layer, storage, mesh) schedule(dynamic)
     for (unsigned int layer_idx = 1; layer_idx < layer_count; layer_idx++)
     {
         Polygons outlines = storage.getLayerOutlines(layer_idx, false);
@@ -978,7 +978,7 @@
         const int max_checking_layer_idx = std::min(static_cast<int>(storage.support.supportLayers.size())
                                                   , static_cast<int>(layer_count - (layer_z_distance_top - 1)));
         const size_t max_checking_idx_size_t = std::max(0, max_checking_layer_idx);
-#pragma omp parallel for default(none) shared(support_areas, storage) schedule(dynamic)
+#pragma omp parallel for shared(support_areas, storage) schedule(dynamic)
         for (size_t layer_idx = 0; layer_idx < max_checking_idx_size_t; layer_idx++)
         {
             support_areas[layer_idx] = support_areas[layer_idx].difference(storage.getLayerOutlines(layer_idx + layer_z_distance_top - 1, false));
```

LipuFei, 2019-08-15:
> Debian has just raised the importance of this issue to "serious": https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=925664
>
> @LipuFei Have you done further successful testing with `default(none)`?
> In [...]

@onitake Changing to `default(none)` works for me with GCC 9, but I haven't done any testing on GCC <=8 yet. I'll do a quick test and let you know.

onitake, 2019-08-15:
Sorry, I meant to say _without_ `default(none)`. :sweat_smile:

LipuFei, 2019-08-15:
> 
> 
> Sorry, I meant to say _without_ `default(none)`. 😅

Without `default(none)` I cannot compile it with GCC 9.1 in MinGW64. I imagine that this will have the same problem with GCC 9.1 on Linux. I can fix this by adding all the shared variables into the `shared()` enclosure, but there's a lot of them... IMO, I think `default(none)` is a better solution in this case.

onitake, 2019-08-16:
@LipuFei With the patch I posted above, I can compile and run just fine with GCC 9.1 and GCC 8.3 on Debian bullseye/sid.
Note that there are minor differences between @hroncok 's patch and mine.

Ghostkeeper, 2019-08-16:
(I'll confirm that Onitake's patch still compiles on GCC 8.3.0 on Ubuntu.)

LipuFei, 2019-08-16:
@onitake Ah, sorry, I got confused... I did exactly the same thing as in your patch, so I can confirm that it also works for GCC 9.1.

onitake, 2019-08-16:
I understand that `default(none)` and a full list of `shared()` variables is the correct way to go, but is there any negative impact from not doing so?

GCC 9 will be the standard at some point, but I think it's too soon to break compatibility just yet.

Another portable solution could be #ifdefs that switch the #pragma between different compilers. But that's not very clean either...

LipuFei, 2019-08-23:
@onitake I agree with you. The compatibility break will probably go on for a long time. We currently have 3 spots which need this fix, so although `#if defined(__GNUC__) && __GNUC__ >= 9` will make it hard to read, it's not everywhere. What do you think? @Ghostkeeper @rburema @nallath

LipuFei, 2019-08-23:
I made a PR for this, with the "right" way. If we want to go for the simple way (remove `default(none)`), it's always easy to do.

Ghostkeeper, 2019-08-23:
I'm not opposed to having preprocessor commands like those in the code if the issue is localised to a certain compiler (as it seems to be in this case).

onitake, 2019-08-26:
Just saw this: https://github.com/gentoo/gentoo/pull/12618#issuecomment-523527979

Looks like clang8 is also affected by the issue.

onitake, 2019-08-27:
Debian has shipped my proposed patch in 3.3.0-3: https://salsa.debian.org/3dprinting-team/cura-engine/commit/60dfa26ef43ec53fa0dcb4d8341b7085a0fc1500
See https://tracker.debian.org/pkg/cura-engine for current build status.

The posted fix with #ifdefs (https://github.com/Ultimaker/CuraEngine/pull/1119) seems to be the way to go for now, and it should work for clang8+ as well. Since Debian is already quite a few versions behind (my fault, sorry folks!), we will wait until this PR lands in a release version before we push the next full Cura update.

Ghostkeeper, 2019-09-02:
Should be fixed for version 4.3.

closed 2019-09-02 by Ghostkeeper

msva, 2019-09-02:
@Ghostkeeper can you also merge #1124 too, please?

Ghostkeeper, 2019-09-03:
Yes, done.