Ultimaker/CuraEngine #989: Thin Raft Adhesion Type

BlackMulch, 2019-02-05:
I'd like to recommend a new adhesion option: the 'thin raft.'

The idea is to add a single layer beneath the model that extension beyond the footprint of the model similar to how a normal raft or brim does. The purpose is simply to enhance adhesion for the entire model with minimum use of material and time. The model would be printed directly on top of the raft layer (no airgap), so removal of the thin raft beneath the model would be difficult or impossible, but the edges (like a brim) could still be trimmed.

This would be useful when printing with materials that tend to have problematic first layers. For example, certain formulations of PETg can be difficult to stick to the bed for the first few to tens of mm in each new extrusion path in the first layer. This new adhesion layer would move these defects to the perimeter of the raft. Then, in a single (or as few as possible) extrusion path cover the build plate area the will sit beneath the model. The next layer (the first of the model) will have a much better chance of success atop this surface.

This is already partially accomplished by the current brim adhesion method, but after printing the brim, each new extrusion path for the model (now interior to the brim, but still directly on the build plate) still has potential to fail to adhere. A raft (as currently available) can also accomplish this, but at a greater cost in material and defects and roughness on the bottom of the model from printing into the airgap and damage from removing the raft. The 'thin raft' could merge these capabilities to provide the adhesion of the raft beneath the entire model along with better bottom surface quality and less post-proceasing of the model.

Ghostkeeper, 2019-02-05:
There is a problem with printing concentric patterns from outside to inside. Quite often the middle part consists of very small circles where the nozzle stays for a fairly long amount of time, causing the nozzle to overheat the surrounding lines. Also if you're slightly overextruding (which is useful for the first layer) then the excess tends to clump up in the middle, causing large bumps of material there which have a chance to detach the print from the bed.

You can sort of achieve this effect by:
* setting Adhesion Type to Brim,
* setting Initial Layer Bottom Pattern to Concentric,
* (optional) enabling Outer Before Inner Walls

In particular the initial layer bottom pattern thing is very effective. But of course concentric patterns print from inside to outside precisely because of the aforementioned reason. Using an older build, like 3.0, could replicate your behaviour exactly because concentric patterns printed from outside to inside then.

BlackMulch, 2019-02-05:
You're exactly right about the pros and cons of this, and the approach you suggest to emulate this behavior does work to some extent.  The problem is that this still leaves the slicer the option to insert a long travel between the final extrusion path of the brim and the start of the base of the model.  The extrusion path following the completion of the brim might be likely to fail to adhere at its start introducing a large defect beneath the model (or a failed first layer altogether).  The 'thin raft' could be tuned to minimize the number and/or length of travels (to preferably a single extrusion path) in the first layer.  Consider a zig-zag surface pattern for the raft:  A single extrusion path to cover the build plate beneath the model (plus a brim), while keeping the most likely sites for first-layer defects and failures (start of extrusion path, extrusion corners, edges, etc.) away from the base of the model.