Ultimaker/CuraEngine #990: [beta 4] [BUG] Unwanted Retract at beginning of a print

hevilp, 2019-02-07:
**Application Version**

**Platform**

**Display Driver**

**Steps to Reproduce**
Slicing an object and retract in the beginning, which I don't want and not configured

G1 F3600 E-4 is unwanted and I'm unable to disable,

**Actual Results**
```
;FLAVOR:Marlin
;TIME:454
;Filament used: 0.273263m
;Layer height: 0.2
;Generated with Cura_SteamEngine 4.0.0-BETA
M82 ;absolute extrusion mode
;start_gcode EXTRUDER=240 BED=100
G92 E0
G1 F3600 E-4
;LAYER_COUNT:2
;LAYER:0
M107
```

Ghostkeeper, 2019-02-07:
Which printer are you using? This looks like it's in your start g-code.

hevilp, 2019-02-07:
No,

my startgcode is the line with start_gcode EXTRUDER=240 BED=100

I'm on a Voron V2.1

Ghostkeeper, 2019-02-07:
Ah, maybe you could attach a project file then so that we can reproduce the problem.

hevilp, 2019-02-07:
3mf file?

[V2_fsr_cali.zip](https://github.com/Ultimaker/CuraEngine/files/2841402/V2_fsr_cali.zip)

Ghostkeeper, 2019-02-07:
No, a project file. That is File -> Save...

This produces a 3MF file that also contains all of your settings.

hevilp, 2019-02-07:
[V2_z_drive_rear_support_b_5mm_x2.zip](https://github.com/Ultimaker/CuraEngine/files/2841421/V2_z_drive_rear_support_b_5mm_x2.zip)

Ghostkeeper, 2019-02-07:
Oh, you're not using a machine definition that is default in Cura. I thought you might be using Custom FFF Printer, but it seems you have a different distribution of Cura than the one we develop, or you modded it in some way? Where did you get your definition for Voron_2_350?

hevilp, 2019-02-07:
No, it is a default Cura Version with ressources files from the Voron community

hevilp, 2019-02-07:
Where does the 4 mm does come from? I got 5 mm retract in my config, not 4 mm

Ghostkeeper, 2019-02-07:
I don't know, but a start code with only a comment line, and a retraction that doesn't use the normal retraction length sounds really suspicious to me :stuck_out_tongue: 

Where did you get those resource files for your Cura mod then? I found a Reddit thread but no definition files there.

hevilp, 2019-02-07:
In the discord community, but I checked the files, there is nothing with the retraction in it.

BTW: my whole start code is in klipper ;) so I only need 1 line in every slicer....

hevilp, 2019-02-21:
I use my own Profile, even here is a retract in the beginning...