Ultimaker/CuraEngine #991: Feature request: Gyroid Support

TimoErdmann, 2019-02-12:
**Application Version**
ALL

**Platform**
ALL

**Actual Results**
actually you can use gradual ZigZag Support to safe some Material and mostly printing time. But nevertheless this method uses an immense amount of time and material because this type of Support is too flimsy. 

**Expected results**
Add Gyroid Support mode. With this option an gyroid infill like pattern is generated with much much less material nedded to generate an equal strenght as the ZigZag support. 
A gradual option would be amazing. But I don't know a way to achive this at the moment.

nallath, 2019-02-14:
For our reference; CURA-6219

smartavionics, 2019-02-15:
I allowed gyroid to be selected for support in my Cura and it looks like it could be usable. I also told it to use 1 support wall line because otherwise the gyroid tends to have very ragged ends. I haven't actually tried printing this yet.

[image]

The actual change I made was trivial, a simple edit to fdmprinter.def.json which you can find (somewhere!) in your Cura installation. I just added the gyroid line in the section shown below...

````
                "support_pattern":
                {
                    "label": "Support Pattern",
                    "description": "The pattern of the support structures of the print. The different options available result in sturdy or easy to remove support.",
                    "type": "enum",
                    "options":
                    {
                        "lines": "Lines",
                        "grid": "Grid",
                        "triangles": "Triangles",
                        "concentric": "Concentric",
                        "zigzag": "Zig Zag",
                        "cross": "Cross",
                        "gyroid": "Gyroid"
                    },
                    "default_value": "zigzag",
                    "enabled": "support_enable or support_tree_enable",
                    "limit_to_extruder": "support_infill_extruder_nr",
                    "settable_per_mesh": false,
                    "settable_per_extruder": true
                },
````

If you edit that file, please be careful as it is very fussy about syntax, an extra or missing comma will stop Cura from starting up.

smartavionics, 2019-02-15:
And with a little more modification of that file you can also allow the Connect Support Lines setting be used with gyroid which is more economical than adding the outline wall...

[image]

```
                "zig_zaggify_support":
                {
                    "label": "Connect Support Lines",
                    "description": "Connect the ends of the support lines together. Enabling this setting can make your support more sturdy and reduce underextrusion, but it will cost more material.",
                    "type": "bool",
                    "default_value": false,
                    "value": "support_pattern == 'cross' or support_pattern == 'gyroid'",
                    "enabled": "support_pattern == 'grid' or support_pattern == 'triangles' or support_pattern == 'cross' or support_pattern == 'gyroid'",
                    "limit_to_extruder": "support_infill_extruder_nr",
                    "settable_per_mesh": false,
                    "settable_per_extruder": true
                },
```

smartavionics, 2019-02-15:
I have created a PR so the devs can evaluate this if they wish.

TimoErdmann, 2019-02-15:
Hey this looks great. I got it to the point with the Support pattern for myself. But the zig-zaggify-option is perfekt. 

Thank you for finding this out and bringing this to a PR.

smartavionics, 2019-02-15:
Hi @TimoErdmann , please try this out if you can and give us some feedback, thanks!

Ghostkeeper, 2019-02-18:
That PR has been merged into Master so you can look forward to it in Cura 4.1.

closed 2019-02-18 by Ghostkeeper