Ultimaker/CuraEngine #993: build failed in windows with mingw32

JockJo, 2019-02-13:
**Application Version**

the latest version
**Platform**

windows 
**Qt**

Qt 5.10
**PyQt**

PyQt5 5.10

**Steps to Reproduce**

At first, i run with cygwin64 and cmake successfully, the command as follow.
```
cmake -DCMAKE_BUILD_TYPE=Release -DCMAKE_PREFIX_PATH=D:\documents\Cura\libArcus\install 
-DCMAKE_INSTALL_PREFIX=../../install -DBUILD_TESTS=OFF -G "MinGW Makefiles ../..

```
then i wanted to make with mingw32 as follow.
`mingw32-make
`
**Actual Results**

then i got the error, how can i solve that.
```
[  1%] Skipping update step for 'stb'
[  2%] No configure step for 'stb'
[  3%] No build step for 'stb'
[  4%] No install step for 'stb'
[  5%] Completed 'stb'
[  8%] Built target stb
[ 10%] Built target clipper
[ 11%] Building CXX object CMakeFiles/_CuraEngine.dir/src/Application.cpp.obj
In file included from D:/Program Files/mingw-w64/x86_64-7.1.0-posix-seh-rt_v5-rev2/mingw64/lib/gcc/x86_64-w64-mingw32/7.1.0/include/x86intrin.h:50:0,
                 from D:/Program Files/mingw-w64/x86_64-7.1.0-posix-seh-rt_v5-rev2/mingw64/x86_64-w64-mingw32/include/winnt.h:1519,
                 from D:/Program Files/mingw-w64/x86_64-7.1.0-posix-seh-rt_v5-rev2/mingw64/x86_64-w64-mingw32/include/minwindef.h:163,
                 from D:/Program Files/mingw-w64/x86_64-7.1.0-posix-seh-rt_v5-rev2/mingw64/x86_64-w64-mingw32/include/windef.h:8,
                 from D:/Program Files/mingw-w64/x86_64-7.1.0-posix-seh-rt_v5-rev2/mingw64/x86_64-w64-mingw32/include/windows.h:69,
                 from D:\documents\Cura\CuraEngine\src\utils/gettime.h:6,
                 from D:\documents\Cura\CuraEngine\src\FffPolygonGenerator.h:11,
                 from D:\documents\Cura\CuraEngine\src\Scene.h:7,
                 from D:\documents\Cura\CuraEngine\src\Slice.h:7,
                 from D:\documents\Cura\CuraEngine\src\Application.h:7,
                 from D:\documents\Cura\CuraEngine\src\Application.cpp:8:
D:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.13.26128/include/mm3dnow.h:19:2: error: #error This header is specific to X86 targets
 #error This header is specific to X86 targets
  ^~~~~
D:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.13.26128/include/mm3dnow.h:27:6: error: #error This header is specific to the X86 target
     #error This header is specific to the X86 target
      ^~~~~
In file included from D:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.13.26128/include/chrono:7:0,
                 from D:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.13.26128/include/mutex:11,
                 from D:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.13.26128/include/google/protobuf/stubs/once.h:34,
                 from D:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.13.26128/include/google/protobuf/message_lite.h:47,
                 from D:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.13.26128/include/google/protobuf/wire_format_lite_inl.h:45,
                 from D:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.13.26128/include/google/protobuf/map_type_handler.h:35,
                 from D:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.13.26128/include/google/protobuf/map.h:49,
                 from D:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.13.26128/include/google/protobuf/generated_message_table_driven.h:34,
                 from D:/documents/Cura/CuraEngine/build/release/Cura.pb.h:26,
                 from D:\documents\Cura\CuraEngine\src\communication/ArcusCommunication.h:11,
                 from D:\documents\Cura\CuraEngine\src\Application.cpp:10:
D:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.13.26128/include/ratio:295:36: error: missing terminating ' character
  uint64_t _Llow = _Lfactor & 0xFFFF'FFFFULL;
                                    ^~~~~~~~~
D:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.13.26128/include/ratio:297:36: error: missing terminating ' character
  uint64_t _Rlow = _Rfactor & 0xFFFF'FFFFULL;
                                    ^~~~~~~~~
[... 26 lines cut ...]
                 from D:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.13.26128/include/string:6,
                 from D:\documents\Cura\CuraEngine\src\Application.cpp:7:
D:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.13.26128/include/vcruntime.h:81:1: error: expected constructor, destructor, or type conversion before '(' token
 _CRT_BEGIN_C_HEADER
 ^
D:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.13.26128/include/vcruntime.h:296:1: error: expected constructor, destructor, or type conversion before '(' token
 _CRT_END_C_HEADER
 ^
D:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.13.26128/include/corecrt.h:1982:1: error: expected constructor, destructor, or type conversion before '(' token
 _CRT_END_C_HEADER
 ^
In file included from D:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.13.26128/include/climits:5:0,
                 from D:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.13.26128/include/xlocnum:6,
                 from D:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.13.26128/include/ios:6,
                 from D:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.13.26128/include/ostream:6,
                 from D:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.13.26128/include/istream:6,
                 from D:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.13.26128/include/string:6,
                 from D:\documents\Cura\CuraEngine\src\Application.cpp:7:
D:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.13.26128/include/yvals.h:153:32: error: expected declaration before end of line
 #pragma pack(push,_CRT_PACKING)
                                ^
mingw32-make[2]: *** [CMakeFiles\_CuraEngine.dir\build.make:71: CMakeFiles/_CuraEngine.dir/src/Application.cpp.obj] Error 1
mingw32-make[1]: *** [CMakeFiles\Makefile2:147: CMakeFiles/_CuraEngine.dir/all] Error 2
mingw32-make: *** [Makefile:151: all] Error 2
```

LipuFei, 2019-02-13:
There is a missing rightside double quote for `"MinGW Makefiles`. From the log it looks like it's still building with MSVC.

JockJo, 2019-02-13:
@LipuFei so sorry, i miss the quote. there is the all command that i used. As follow.
```
$ cmake -DCMAKE_BUILD_TYPE=Release -DCMAKE_PREFIX_PATH="D:\documents\Cura\libArcus\install" -DCMAKE_INSTALL_PREFIX="../../install" -DBUILD_TESTS=OFF -G "MinGW Makefiles" ../..
-- The C compiler identification is GNU 7.1.0
-- The CXX compiler identification is GNU 7.1.0
-- Check for working C compiler: D:/Program Files/mingw-w64/x86_64-7.1.0-posix-seh-rt_v5-rev2/mingw64/bin/gcc.exe
-- Check for working C compiler: D:/Program Files/mingw-w64/x86_64-7.1.0-posix-seh-rt_v5-rev2/mingw64/bin/gcc.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: D:/Program Files/mingw-w64/x86_64-7.1.0-posix-seh-rt_v5-rev2/mingw64/bin/g++.exe
-- Check for working CXX compiler: D:/Program Files/mingw-w64/x86_64-7.1.0-posix-seh-rt_v5-rev2/mingw64/bin/g++.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building with Arcus
-- Found Protobuf: D:/cygwin64/lib/libprotobuf.lib (found suitable version "3.6.1", minimum required is "3.0.0")
-- Building Stb from source.
-- Created Stb installation at: D:/documents/Cura/CuraEngine/build/release/stb-prefix/src
-- Compile with more optimization flags
-- Found OpenMP_C: -fopenmp (found version "4.5")
-- Found OpenMP_CXX: -fopenmp (found version "4.5")
-- Found OpenMP: TRUE (found version "4.5")
-- Using windres
-- Configuring done
-- Generating done
-- Build files have been written to: D:/documents/Cura/CuraEngine/build/release

JockJo@DESKTOP-3MFE8QH /cygdrive/d/documents/Cura/CuraEngine/build/release
$ mingw32-make
Scanning dependencies of target stb
[  1%] Creating directories for 'stb'
[  2%] Performing download step (git clone) for 'stb'
Cloning into 'stb'...
Your branch is up-to-date with 'origin/master'.
Already on 'master'
[  3%] No patch step for 'stb'
[  4%] Skipping update step for 'stb'
[  5%] No configure step for 'stb'
[  6%] No build step for 'stb'
[  7%] No install step for 'stb'
[  8%] Completed 'stb'
[  8%] Built target stb
Scanning dependencies of target clipper
[  9%] Building CXX object CMakeFiles/clipper.dir/libs/clipper/clipper.cpp.obj
[ 10%] Linking CXX static library libclipper.a
[ 10%] Built target clipper
[ 11%] Running cpp protocol buffer compiler on Cura.proto
Scanning dependencies of target _CuraEngine
[ 12%] Building CXX object CMakeFiles/_CuraEngine.dir/src/Application.cpp.obj
In file included from D:/Program Files/mingw-w64/x86_64-7.1.0-posix-seh-rt_v5-rev2/mingw64/lib/gcc/x86_64-w64-mingw32/7.1.0/include/x86intrin.h:50:0,
                 from D:/Program Files/mingw-w64/x86_64-7.1.0-posix-seh-rt_v5-rev2/mingw64/x86_64-w64-mingw32/include/winnt.h:1519,
                 from D:/Program Files/mingw-w64/x86_64-7.1.0-posix-seh-rt_v5-rev2/mingw64/x86_64-w64-mingw32/include/minwindef.h:163,
                 from D:/Program Files/mingw-w64/x86_64-7.1.0-posix-seh-rt_v5-rev2/mingw64/x86_64-w64-mingw32/include/windef.h:8,
                 from D:/Program Files/mingw-w64/x86_64-7.1.0-posix-seh-rt_v5-rev2/mingw64/x86_64-w64-mingw32/include/windows.h:69,
                 from D:\documents\Cura\CuraEngine\src\utils/gettime.h:6,
                 from D:\documents\Cura\CuraEngine\src\FffPolygonGenerator.h:11,
                 from D:\documents\Cura\CuraEngine\src\Scene.h:7,
                 from D:\documents\Cura\CuraEngine\src\Slice.h:7,
                 from D:\documents\Cura\CuraEngine\src\Application.h:7,
                 from D:\documents\Cura\CuraEngine\src\Application.cpp:8:
D:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.13.26128/include/mm3dnow.h:19:2: error: #error This header is specific to X86 targets
 #error This header is specific to X86 targets
  ^~~~~
D:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.13.26128/include/mm3dnow.h:27:6: error: #error This header is specific to the X86 target
     #error This header is specific to the X86 target
      ^~~~~
In file included from D:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.13.26128/include/chrono:7:0,
[... 45 lines cut ...]
 _CRT_BEGIN_C_HEADER
 ^
D:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.13.26128/include/vcruntime.h:296:1: error: expected constructor, destructor, or type conversion before '(' token
 _CRT_END_C_HEADER
 ^
D:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.13.26128/include/corecrt.h:1982:1: error: expected constructor, destructor, or type conversion before '(' token
 _CRT_END_C_HEADER
 ^
In file included from D:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.13.26128/include/climits:5:0,
                 from D:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.13.26128/include/xlocnum:6,
                 from D:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.13.26128/include/ios:6,
                 from D:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.13.26128/include/ostream:6,
                 from D:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.13.26128/include/istream:6,
[... cut ...]
D:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.13.26128/include/yvals.h:153:32: error: expected declaration before end of line
 #pragma pack(push,_CRT_PACKING)
                                ^
mingw32-make[2]: *** [CMakeFiles\_CuraEngine.dir\build.make:71: CMakeFiles/_CuraEngine.dir/src/Application.cpp.obj] Error 1
mingw32-make[1]: *** [CMakeFiles\Makefile2:147: CMakeFiles/_CuraEngine.dir/all] Error 2
mingw32-make: *** [Makefile:151: all] Error 2

```

and the other dependency(arcus, protobuf)is build with mscv and nmake. is there conflict when i used different build tools to build ?

Ghostkeeper, 2019-02-14:
You need to build Arcus and Protobuf with MinGW as well in order to link with CuraEngine. And you'd need to build Arcus and Protobuf a second time with MSVC in order to link it to CPython.

The errors you're seeing are happening because your build seems to pull in headers from MSVC (MinGW points out header files in the visual studio installation folder). I don't know why it's doing that but you may want to clear your CMake cache or otherwise look at your `%PATH%`.

JockJo, 2019-02-14:
do you mean that i need to build twice time with  mingw  and  MSVC?

Ghostkeeper, 2019-02-14:
To create a full build of Cura on Windows, you need to build libArcus and Protobuf twice: Once with MinGW (to link to CuraEngine) and once with MSVC (to build libArcus into a Python module).

LipuFei, 2019-02-14:
Hi @JockJo not sure what happened. I'm using msys2 with mingw64 and I haven't encountered this before. Could you do a `mingw32-make --trace` and parse the output? Thanks.

JockJo, 2019-02-14:
thanks, i will try it  right away

city8208, 2019-02-24:
Hi
I install Mingw64 today.
But the install exe got some trouble.

And they are not sure what time they can fix.
How shoud I install on windows in other way

JockJo, 2019-02-24:
@city8208 do you mean that install mingw64 failed?

there is the introduction.https://gist.github.com/ChrisTerBeke/cd943274cc5275ff80c724260e848c8d