Ultimaker/CuraEngine #998: force to Z hop / retractation  instead of G0

jlmalet, 2019-02-20:
**Application Version**
cura 3.6

**Platform**
windows

**Display Driver**
not relevant

**Steps to Reproduce**
create an object with a 4cm x 4cm flat horizontal part
activate retractation settings 
print the object

**Actual Results**
cura engine generate lot of G0 commands, this create some nasty visual effect (some "scatch") on the flat layer, and more on my printer this can lead to have the head bump on printed parameters (for example if there is a perimeter included into another, and that cura tries to print the internal of both then the outside, and begins with the outer perimeter the head will bump into the internal wall of the outer parameter when comming back from the internal permiter.... on my printer, this can create head position computation mistake and even create 1cm shift.... I don't want to have G0 created, instead use Zmove, XY G0 move, Zmove and then use retractation 

**Expected results**
no scratch so no actual print when moving, no head bump, no colision... 

**Additional Information**
would like to have "use zhop instead of fast move everywhere" option

smartavionics, 2019-02-20:
Set combing to "no skin" and set the Max Comb Distance Without Retract to something like 10. You can also tell it to z-hop as well.

Ghostkeeper, 2019-02-21:
Or to "Off" if you also want it to retract when moving through infill.

I think that answers the question.

closed 2019-02-21 by Ghostkeeper