Ultimaker/CuraEngine #1001: Feature Request: Print Order ABAB for small Areas / print islands sequentially

TimoErdmann, 2019-02-25:
**Application Version**
ALL

**Platform**
ALL

**Actual Results**
In a mulitmaterial Print Cura will print Material A and Material B in the following sheme: 
AABB 
This results in overheating areas if the print time of these islands or layers are too short. The material has not enough time to transfer its heat in the bottom layers or to the surrounding.
The same is happening for single material prints. If there are more than one island, sometimes the layer will end with a small island and the next layer will start at the same. So if this island is to short it will overheat, like the multi material sheme.

**Expected results**
There need to be two options. One for Multimaterial prints and another one for singlematerial prints.
1. Multimaterial prints:
Add an option to print the Materials in an alternating order like: ABAB 
Yes the actual printing time will increase, but the overheating will be minimized.

2. single Material prints or for Multimaterial prints with AABB sheme:
if the area of one island is smaller than X mm^2: Print the islands sequentially. 
so if there are three islands: 1,2 and 3. If island 2 is smaller than X mm^2 print with this sheme: 1,3,2,1,3,2
otherwise if no island is smaller than X  print with the current sheme: 1,2,3,3,2,1

Ghostkeeper, 2019-02-25:
If you're experiencing overheating, I suggest you increase the Minimum Layer Time so that it waits for the layer to cool down. I'd wager that will be more efficient than to switch extruders in order to do another part first.

Your proposal has three problems, one process-related and two technical, that make it pretty improbable for it to be implemented.

1. Process-related: When switching extruders multiple times per layer, you will need to prime multiple times per layer (for some materials). So you'll need multiple prime towers. This is going to have a great negative effect on your printing time.
2. Technical: CuraEngine slices multi-threaded, with each layer being processed in a different thread. This means that the thread that determines the order for one layer doesn't know what order the previous layer printed the parts in.
3. Technical: It's unknown whether one polygon on layer X is the same part as a polygon on layer X+1. It's rare that all of the vertices are the same, and quite often some polygons will overlap with multiple polygons on the next layer (such as when a part splits in two). So the scheme of 1,2,3,3,2,1 is hard to achieve: that first 2 is not the same as the second 2.

TimoErdmann, 2019-02-25:
Thank you for your reply. 

Ok. I understand your points. 
for point 2: So it is completly random which island will print first? Is there a way to sort those individual islands in a descending way denpending on their print time? 
So if ***island 1*** needs 3 seconds, ***island 2*** needs 20 seconds and ***island 3*** 40 seconds. In the next layer printing time is as following: 
***island 1*** 4 seconds
***island 2*** 35 seconds
***island 3*** 33 seconds
The printing order will be: 3,2,1,2,3,1
This way the smallest island has probably enough time to cool a enough. I think the printing time is a good characteristic to identify those islands. Defenitly this will not match every possibility but it covers a large number.

Yes minimum layer time will fix this issue but it adds unnecessary printing time. For a 6 hour print this is not a big problem. But i will print 24+ h prints frequently. If this will save only 1 hour it is a great benefit.

Point 1:
Yes swichting the extruder many times per layer isn't the proper way. But is it possible to change order of printing the layers? 
If we have a dual extruder setup. ***A*** is extruder one and extruder ****B*** the second. At the moment the printing order is: AABB 
So extruder ***A*** will end the layer and start the next one. the extruder is changed to ****B*** Now extruder ***B*** finishes this layer and start with a new one. So this pattern is AABB

Is it possible to change this behavior to ABAB:
Extruder ***A*** starts layer 1, extruder is switched to ***B***, extruder ***B*** finishes Layer1
In the next layer:
switch to Extruder ***A***, extruder ***A*** starts this layer, switch to exturuder ***B***, extuder ***B*** finishes this layer

Ghostkeeper, 2019-02-27:
> So it is completly random which island will print first?

No, it does this by minimising travel moves, but minimising travel moves depends on the location within a part where it finished printing that part. So it's going to generate all moves necessary to print a part, all walls, skin, infill lines, etc. and also optimising the order of these infill lines to minimise travel moves. And then it's going to stand at the final location of the last line and see which part is now the closest. You can't do this beforehand.

Also to know the printing time, you'd need to generate all of the lines and know the order in which they are going to be printed, and you really want that to be done multi-threaded. I think surface area is an easier metric to work with.

> Is it possible to change this behavior to ABAB:
> Extruder _**A**_ starts layer 1, extruder is switched to _**B**_, extruder _**B**_ finishes Layer1
> In the next layer:
> switch to Extruder _**A**_, extruder _**A**_ starts this layer, switch to exturuder _**B**_, extuder _**B**_ finishes this layer

As you already counted this results in two extruder switches per layer instead of the current one switch per layer.