Ultimaker/CuraEngine #1006: Filament used with multiple extruders is not calculated

marianopaulin02, 2019-03-05:
**Application Version**
4.0 (master)

**Platform**
linux x64

**Steps to Reproduce**
$ ./CuraEngine slice -v -j fdmprinter.def.json -o output.gcode -e0 -l some.stl -e1 -l somestl1.stl

**Actual Results**
Only shows filament used for the first extruder, second one is 0.
(neither is the sum of both extruders)

**Expected results**
show the filament used for each extruder

**Additional Information**
Maybe related with #930  
I think it's something related with CommandLine.cpp: last_extruder is never updated
ExtruderTrain& last_extruder = slice.scene.extruders[0];
Not working adding -s extruder_nr= as mentioned in #403  

I've been analyzing the problem for days, thank you in advance.

Ghostkeeper, 2019-03-06:
Indeed, I could reproduce this bug. Though I also had to add `-j fdmextruder.def.json` to make it slice.

marianopaulin02, 2019-03-06:
Thank you for the reply. Yes I forgot it, I 've modified ``fdmprinter.def.json`` `to add the second extruder, but nevermind, let's use ``ultimaker3.def.json`` instead. Do you achieve the expected result? or have any idea of how to solve the bug?

Ghostkeeper, 2019-03-07:
It's not just that it doesn't record the material usage. It's that it doesn't seem to place the object on the second extruder. The resulting g-code has no `T1` command in it.

LipuFei, 2019-03-08:
JIRA ticket: CURA-6312

Ghostkeeper, 2020-12-30:
This seems to have been fixed in the meanwhile. I can no longer reproduce the issue.

A couple of commits were made a while ago that were apparently aimed at fixing this issue, although not through a pull request:
* https://github.com/Ultimaker/CuraEngine/commit/6a52bb1261fb28c63123e8388a08af6593ba561c
* https://github.com/Ultimaker/CuraEngine/commit/f1c06d5c71f4db241736824647473548c517aa4e
* https://github.com/Ultimaker/CuraEngine/commit/d6043cf7be00bd16c2940c7f3b3a8082e18d0d37

closed 2020-12-30 by Ghostkeeper