Ultimaker/CuraEngine #1007: Why "Adaptive Layer Height Threshold" is rounded up to 360?

alexjx, 2019-03-08:
From the code, it seems cura engine is expecting the threshold to be an angle value.
However, the algorithm doesn't.
 
https://github.com/Ultimaker/CuraEngine/blob/24e57cbd6b215ca21b4cfb31ca474dec600042a9/src/FffPolygonGenerator.cpp#L126

This actually limited cura engine to find a more faster solution for a model, since the threshold will be capped at 360.

**Application Version**

3.6

**Platform**

Windows

**Qt**

**PyQt**

**Display Driver**

**Steps to Reproduce**

Change "adaptive layer height threshold to 300 and 400. 
Slice the model accordingly.

**Actual Results**

Cura uses more aggressive layer height (thinner) when setting to 400 than 300.

**Expected results**

400 should relax the layer height as what it did from 0 to 360.

**Additional Information**

Ghostkeeper, 2019-03-08:
Seems like this was misinterpreted somewhere along the way. The threshold value is compared to the tangent of the surface angle, so it would be a distance rather than an angle. It should not even be floating point, but just a distance in mm which gets converted to `coord_t` in the engine. Good find.

alexjx, 2019-03-10:
Thanks for confirming it. I'm trying to understanding the algorithm. From my understanding, what the threshold compares to is the ratio of (layer height / tangent of the slop). Why it should be treated as "distance"?

Ghostkeeper, 2019-03-11:
Sorry, I was mistaken last Friday. I was also looking at that ratio and thinking it was referring to [this formula](https://github.com/Ultimaker/CuraEngine/blob/master/docs/assets/support_distance_formula.svg) that we are using to calculate the overhang distance for support. But it seems that this is something different (`/` instead of `*`).

Ghostkeeper, 2019-03-11:
Fixed here: https://github.com/Ultimaker/CuraEngine/commit/99ee78237b8d356ccc6096a2575fa4d73e15d166

closed 2019-03-11 by Ghostkeeper