Ultimaker/CuraEngine #1018: How can I use the indicated protobuf lib?

thzhao, 2019-03-26:
I installed the protobuf in my home directory. There are two protobufs with different compiler in my system. It seems the link error.  How can I indicate the protobuf I want?

[100%] Building CXX object CMakeFiles/CuraEngine.dir/src/main.cpp.o
Linking CXX executable CuraEngine
lib_CuraEngine.a(commandSocket.cpp.o): In function `google::protobuf::internal::GetEmptyStringAlreadyInited[abi:cxx11]()':
/usr/local/include/google/protobuf/message_lite.h:134: undefined reference to `google::protobuf::internal::fixed_address_empty_string[abi:cxx11]'
lib_CuraEngine.a(commandSocket.cpp.o): In function `cura::Listener::error(Arcus::Error const&)':
commandSocket.cpp:(.text._ZN4cura8Listener5errorERKN5Arcus5ErrorE[_ZN4cura8Listener5errorERKN5Arcus5ErrorE]+0x36): undefined reference to `Arcus::Error::toString[abi:cxx11]() const'
commandSocket.cpp:(.text._ZN4cura8Listener5errorERKN5Arcus5ErrorE[_ZN4cura8Listener5errorERKN5Arcus5ErrorE]+0x75): undefined reference to `Arcus::Error::toString[abi:cxx11]() const'

rburema, 2019-03-27:
@thzhao Did you do the configuration with CMake? Yous should be able to specify where protobuf is build in the CMake parameters.

LipuFei, 2019-08-23:
@thzhao You can use certain cmake arguments such as `CMAKE_PREFIX_PATH` to make CMake look for where your libraries are installed first. See documentation: https://cmake.org/cmake/help/v3.12/variable/CMAKE_PREFIX_PATH.html

You can have a look at the links for `find_package()`, `find_library()`, etc. They explain how those functions behave, where do they look for things first.