Ultimaker/CuraEngine #1019: Different result with CuraEngine and Cura app

serhattsnmz, 2019-03-26:
**Application Version**
CuraEngine 3.6

**Platform**
All platform

**Steps to Reproduce**
I am trying to integrate CuraEngine in my project. The problem is I am getting different result than Cura Desktop app. 

I am running CuraEngine as;
`CuraEngine slice -v -j /cura/resources/definitions/ultimaker3_extended.def.json -o /test.gcode' -l "test.stl"`

For example the result is 149951 mm3 ( ~185 gr ) in CuraEngine. When I try to slice same model in Cura app, it says 256gr. 

I cannot find the real problem and how to make them equal.

rburema, 2019-03-27:
Hi @serhattsnmz  Cura puts the command-line it uses to initialize a run of the engine in the log, you can find it if you search the log for lines printed by the backend.

serhattsnmz, 2019-03-27:
@rburema Where I can find the logs?

rburema, 2019-03-27:
Depends on your OS. You can look [here](https://github.com/Ultimaker/Cura/wiki/Cura-Preferences-and-Settings-Locations) to check.

serhattsnmz, 2019-03-27:
Thanks for help @rburema. I've found the logs and tried to slice again with same parameters, but problem is still same - different results and different gcodes. Might Cura have any other process before or after CuraEngine which effect gcode result?

rburema, 2019-03-28:
@serhattsnmz Unless the post-processing plugin does something, no. Are you using the CuraEngine '3.6'-branch/tag? Are the dependencies (from Ultimaker) of that same branch/tag as well?

(Keep in mind that the parameters from the log might be broken up into multiple lines as well.)

serhattsnmz, 2019-03-28:
Yeah, I'm using CuraEngine 3.6 and all dependencies are set.
Can I use .cfg files in `resources/quality/...` with CuraEngine? If I can, how?

Ghostkeeper, 2019-03-28:
No, CuraEngine is not able to read .cfg setting files.

For us it's really hard to say what is different in your two runs :disappointed:

serhattsnmz, 2019-03-28:
Thanks for reponses guys. 
I really don't know what I'm doing wrong. Here is what I'm doing : 

### Step1:
Slice model with CuraEngine :
```
CuraEngine slice -v -j /cura/definitions/ultimaker_original.def.json -s layer_height=0.1 -o /test/output.gcode -l test.stl
```
Result:
```
Starting inset+skin...
Layer count: 433
Progress: inset+skin accomplished in 0.243s
Starting support...
Processing platform adhesion
Progress: support accomplished in 0.141s
Starting export...
Multithreading GcodeLayerThreader with 2 threads.
Progress: export accomplished in 1.064s
Total time elapsed  1.47s.
Gcode header after slicing:
;FLAVOR:Marlin
;TIME:28632
;Filament used: 8.18679m
;Layer height: 0.1
End of gcode header.
Print time (s): 28632
Print time (hr|min|s): 7h 57m 12s
Filament (mm^3): 52226
```

> **Note:** The definition is stored under `C:\Program Files\Ultimaker Cura 4.0\resources\definitions` which Cura Desktop use.

> **Note:** CuraEngine's version is 3.6. You can take a loot at my enviroment with this link : https://github.com/serhattsnmz/cura-engine-env

### Step 2:
Open Cura >> Select printer : Ultimaker Original >> Set default Values "Fine (0.1mm)"

Result: 
- 29 g
- 3.71 m
- 4 hr 6 min

> **Note:** Cura's version is 4.0, but I also tested with 3.6 and results were same.

As you guys see, they give different result than each other.
I do not why and how to make them equal.

Ghostkeeper, 2019-03-29:
I think the difference is in that you don't supply any material settings when calling CuraEngine.

Cura's front-end has a complex settings system with not only the printer definition, but also the machine changes (customised settings), machine variants (e.g. has a heated build plate or not), global quality profiles, custom global profiles, custom global settings, extruder definitions, nozzle profiles, materials, per-extruder profiles specialised for materials and nozzles, custom per-extruder profiles and custom settings. In theory all of these would have to be applied in order to get the exact same result as in the front-end. In practice, for your Ultimaker Original, most of them are probably empty.

serhattsnmz, 2019-04-03:
All informations you gave me help a lot. Thank you guys.

closed 2019-04-03 by serhattsnmz