Ultimaker/CuraEngine #1028: Draft shield on raft

rlarge, 2019-04-07:
I have be messing around for a bit with the draft shield and have noticed when printed on a raft thea "raft air gap" is also used.  It would be better if it could set the first layer height of the draft shield. 

If this doesn't work this way then just ignore this comment. lol

It would also be cool if it was possible a multi shell/wall at the bottom to one shell/wall at the top. Or maybe a dome setting lol

Thanks

rburema, 2019-04-10:
The way it works now internally is that we just put the raft underneath the entire rest of the slices, so 'layer 0' in this case would be the first layer on the raft. I won't exactly be easy to restructure this so that your suggestion works.

Ghostkeeper, 2019-04-11:
We're going to defer this issue. It's indeed something that could be improved but draft shield is not used very often (and it's experimental).

Timocop, 2020-01-13:
Any updates on this? I wanted to use the draft shield as a prime runway method for each layer because im printing very small perimeters sometimes with retraction (even with super small retraction distances) that causes clogs. I wanted to test if this trick could recover the pressure in the nozzle instead of turning the temp up.
Also this would be usefull for ABS and other filament that can warp easy and need an enclosure to keep it warm.

Wouldnt be this an easy fix? Just ignore the air gap?

Ghostkeeper, 2020-01-14:
There's no update in this. We see the use but the userbase for this feature is too small to get our priority.

We can't just ignore the air gap since there are no layers in the air gap. We need a similar solution to how we're printing support on top of rafts.

Timocop, 2020-01-15:
Aww ok :(