Ultimaker/CuraEngine #1031: [Cura 4.0] Top/Bottom Line pattern collides with nozzle on Comb-Travel

Timocop, 2019-04-08:
Because the top/bottom line pattern makes a travel move at the end of each line. The ozzed material keeps getting stuck at the nozzle while the travel move which causes to warp/curl upwards if they are printed in mid-air (if the top/bottom wall is not reachable, which happens on curves). Thus, the nozzle keeps colliding with the warped edges on combing moves.
However, this doesnt happen with the zig-zag pattern because it pushes the material down all the time even when printing in mid-air causing no warping/pulling.

Its even worse when combing is set to "Within infill" or "Not skin". Cura will go around the skin, but there is no 'Travel Avoid Distance' so it will keep the nozzle on the edges sandpapering the nozzle. The whole printer will vibrate on high travel speeds.

[image]
Once the travel move happens the previous line will get stuck to the nozzle and warp/curl and the second line will drop:
[image]
Due to constant extrusion the mid-air line will drop instead of warp/curl:
[image]

I came back from Slic3r PE, it has Zig-Zag pattern by default. I never ran into this problem there after days of big prints. However, the first print with Cura again and my printers nozzle keeps crashing into curled skin lines. And if Combing 'Within infill' or 'Not skin' is enabled it will sandpaper over skin edges.

**Application Version**
Cura 4.0 (and below)

**Platform**
Windows 10 x64

**Steps to Reproduce**
Optional: Set combing to 'Within infill' or 'Not skin'.
Set infill higher so the top/bottom layer are bearly connect to infill (to create some mid-air top/bottom layers)

**Actual Results**
Nozzle colliding with warped/curled top/bottom line pattern edges.

**Expected results**
The nozzle should not travel on top/bottom edges. It should take 'Travel Avoid Distance' into account. Also Z-Hop (if enabled) when starting top/bottom layers (Or make 'Combing Within Infill' default).

**Additional Information**
This behaviour only seems to happen when the end-line is longer than the start-line. Which pulls back the end-line edge while travel and curles up.

smartavionics, 2019-04-08:
Hello @Timocop , you may be interested in a new feature I have implemented that (I think) solves this problem. It is briefly described in https://github.com/Ultimaker/Cura/pull/5595. If you use Cura on Linux or Windows you can try this feature by installing one of my releases which can be found at https://www.dropbox.com/sh/s43vqzmi4d2bqe2/AAADdYdSu9iwcKa0Knqgurm4a?dl=0

Timocop, 2019-04-09:
That looks pretty neat indeed! Thats what i love about the Cura community. But the Zig-Zag pattern fixes the problem so i switched to that instead. The only problem i have is that the Line Skin Pattern is default.  Which may cause problems for all users using Cura. Not sure why the Zig-Zag pattern is not default?

Ive been using Cura since i have a 3d printer. And i always kept running into this "nozzle crashing into warped skin edges" problem.

rburema, 2019-04-10:
It's good that I see this, I was just about to go to (the) materials (department) to ask if they can (also) see a use for @smartavionics new PR :-)

rburema, 2019-04-11:
We're going to review/test and hopefully merge the mentioned PR (materials dept. said yes, please), and you have a workaround. Can we consider this issue closed then?

Timocop, 2019-04-11:
While #5595 minimizes the effect of warping/curl because its infill. Its still happening on loose edges at curves. Escpecialy if you use easly warpable materials like ABS which reaaaly loves to warp upwards.

I dont realy like the workaround because i want to use Lines, not Zig-Zag. Zig-Zag pattern already has its own problems. I think its better to avoid infill (even the outer wall) when having Combing to 'Within infill' or 'Not in Skin'. Its obviously traveling inside the infill shown in the first picture. Can we make it 'Travel avoid distance' aware?

smartavionics, 2019-04-11:
> While Ultimaker/Cura#5595 minimizes the effect of warping/curl because its infill. Its still happening on loose edges at curves. Escpecialy if you use easly warpable materials like ABS which reaaaly loves to warp upwards.

You have tried it out and that is what you observed?

>  I think its better to avoid infill (even the outer wall) when having Combing to 'Within infill' or 'Not in Skin'. Its obviously traveling inside the infill shown in the first picture. Can we make it 'Travel avoid distance' aware?

I should think it would be possible for the no-skin combing path to move into the infill. I will look into that.

smartavionics, 2019-04-11:
> I should think it would be possible for the no-skin combing path to move into the infill. I will look into that.

What I meant was it should be possible for the no-skin combing path to move into the infill where it is on the boundary of skin (rather than where the infill meets a wall).

Timocop, 2019-04-11:
Yes, i tried your version. It sure was better! But on cuves the edge infill just drops into the model anyways (escpecialy with gyroid). No bound, started warping. It pulled the end of the line to the beginning of the other line. Which makes the edge uneven. Which also collides with the nozzle.

Im printing with a 0.6mm nozzle at 0.3 layer hight. I had worse experiance with smaller nozzles (0.3mm) and low layer hights. After more than half year printing with the same nozzle, Cura has grinded the tip off due all this skin traveling at 120mm/s (even with 'Within infill') on large prints. My 3d printer sounded more like a CNC drill machine when it started printing skin layers.

Offtopic: Ive been testing Slic3r PE recently, i never ran into the uneven skin problem. Even with ABS. Slic3r makes a thicker and wider zig-zag skin layer first. Which acts more like a foundation. Instead of a skin wall which can be too thin, warp, deform or go under the skin layer and push the skin upwards in the early layers.

Instead of wasting material on edge loops (which mostly wont work on curves aynways) why not make a first top skin layer with a larger surface area (dynamicly adjusted to the infill desnity maybe) that acts like a foundation?
Concentric as first top layer (creating a safty-net like foundation), Zig-Zag for middle and Lines (for better surface quality) as top. Doesnt seem to be that complicated. In know there is 'Skin Expand Distance' but it expands every layer which is kinda wastefull in my opinion.
Maybe ill add a feature request for that soon.

And youre right, 'No in Skin' should be an exception. But 'Within infill' should keep the nozzle away from the skin edges. Its not infill area.

Apologies for grammar errors. Im not a native english speaker.

smartavionics, 2019-04-11:
Firstly, no need to apologise for your grammar, it's probably better than mine and I am a native English speaker.

So, I think it would be helpful here if you could provide a cura project file (.3mf) that illustrates the problem for me to study. Thanks.

Timocop, 2019-04-11:
Sure thing.
[PI3_BodyLow.zip](https://github.com/Ultimaker/CuraEngine/files/3070601/PI3_BodyLow.zip)

smartavionics, 2019-04-12:
Hello @Timocop , I have just posted a new build that routes travels on the infill side of an infill/skin boundary. Please see if it fixes your issue.

Timocop, 2019-04-26:
Havnt tried it yet but looks realy good in the virtualizer! I cant print right now because i have no filament. I will test it in depth when i get some. Probably in 1-2 weeks. Not sure about the hardcoded distance part. People may want to be able to change it. Thanks for the effort! :)

smartavionics, 2019-04-26:
Cura already has far too many options, I am now very reluctant to add more.

Timocop, 2019-04-26:
Thats fair.