Ultimaker/CuraEngine #1032: [Cura 4.0] Some Infill Pattern ignore 'Infill Overlap Percentage'

Timocop, 2019-04-11:
Infill Pattern that ignore 'Infill Overlap Percentage':
 - Cubic Subdivision
 - Cross
 - Cross 3D
 - Gyroid

If this is intended behavior, why not hide 'Infill Overlap Percentage' like when 'Concentric' infill pattern is selected?

**Application Version**
Cura 4.0

**Platform**
Windows 10 x64

**Steps to Reproduce**
Select one of aboves infill. Set it more than 5% infill overlap. Notin diff. Wat givs?

Cubic works as intended:
[image]
Gyroid doesnt:
[image]

Timocop, 2019-04-11:
Seems like @smartavionics Cura version fixed this issue. I guess its time to switch.
His comment and release link can be found here: https://github.com/Ultimaker/CuraEngine/issues/1031#issuecomment-481013028

smartavionics, 2019-04-11:
The fix for gyroid should be in the next Ultimaker release.

rburema, 2019-04-12:
... and another two lines changed for the other 3 cases. Made a PR, #1036 my colleages may (or may not) pick it up on monday :-)

closed 2019-12-23 by Timocop