Ultimaker/CuraEngine #1034: Infill pattern to bond vertically across layers [ver 3.6.0]

bobbyfiend, 2019-04-11:
**Application Version**
[3.6.0]

**Platform**
Any

**Expected results**
Feature request: an infill pattern that bonds multiple layers together vertically. I don't know if this is feasible, but maybe a 100% infill option where the filled area would actually look like swiss cheese, with a matrix of vertical holes, aligned 2, 3, or more layers deep, same diameter (roughly) as the nozzle diameter (e.g., 0.4mm). After, for example, 3 layers of solid fill were printed with these 3-layer-deep holes in them, aligned,  the print head would go to each hole and squish a hot burst of PLA down into the holes, vertically bonding the three layers together, potentially more strongly than some layer bonds.

As suggested by /u/jediminer543 (see [here]([url](https://www.reddit.com/r/3Dprinting/comments/bbz8kt/question_is_there_a_slicer_algorithm_to_create/))), holes could be staggered across layers, so one set of holes bonded layers 1, 2, and 3 together, while another set bonded 2, 3, and 4, and another 3, 4, and 5, etc.

Is this feasible? It might make very strong parts.

Mostly I don't need this, but for parts that need to withstand stress on 2 or 3 axes, maybe it would increase the final strength a little? Does a slicer offer anything like it?

**Additional Information**
Here is an ASCII diagram of what I'm imagining: a cross-section in which hyphens represent regular solid layers and pipes (|) represent the vertical squished-in "plugs" that might bond layers together:

     -- -----|-- -----|-- -----|-- -----
     -----|--|-----|--|-----|--|-----|--
     --|--|--|--|--|--|--|--|--|--|--|--
     --|--|-----|--|-----|--|-----|--|--
     --|-----|--|-----|--|-----|--|-----
     -----|--|-----|--|-----|--|-----|--
     --|--|--|--|--|--|--|--|--|--|--|--
     --|--|-----|--|-----|--|-----|--|--
     --|--------|--------|--------|-----

Ghostkeeper, 2019-04-12:
You can achieve this effect by setting the Infill Line Multiplier to 3 (or so) and selecting an infill pattern that staggers, such as Quarter Cubic, Gyroid or even just Lines. Try that, perhaps with Quarter Cubic first, and see if you like that effect.

bobbyfiend, 2019-04-12:
I'm definitely going to try this. Thanks.

AbeFM, 2019-08-23:
What is that doing beyond making each 2D layer of support wider? It doesn't sound like it's doing what he asked (regardless of what he asked being reasonable. :-) I do like the idea, I'm not sure how well it would work in practice.
Perhaps draw a part that is what you want and manually put in the lines to do the fills?

Perhaps the OP just needs to turn down infill speeds or turn up infill line widths to improve adhesion?
What are you trying to accomplish?

Ghostkeeper, 2019-08-25:
In the spirit of OP's post, I understood that he intended to have an infill pattern where the lines would be moved closer together horizontally in order to allow the lines on the next layer (that are shifted slightly according to the pattern) to stick better. Multiplying infill lines doesn't shift lines, but only places new lines. But it has the same effect.

AbeFM, 2019-08-25:
I was assuming he meant something like this:
[image]

Leaving vertical holes and then squirting plastic multiple layers down to bond it all together.

bobbyfiend, 2019-08-25:
Thanks, AbeFM, that's pretty much exactly what I had in mind, except that my imaginings filled in the gaps between the vertical red sections in each layer, though I don't know how important that would be. That diagram is much better than my bad ASCII art, and gets the idea across beautifully.

As for my goal... IDK? I just thought I might have hit on a novel way of bonding layers together more strongly, which could be applied to a huge number of projects. It seems incredibly unlikely that I would be the first person to think of this, so I wanted to put it out there so whoever really thought of it first might see it, and so that smarter people could suggest reasons it wouldn't work, ways to make it better, things that are already doing the same thing, etc. In this thread at least some of that is already happening.

BagelOrb, 2019-10-29:
It's called 'Z pinning'. Here's a paper about it:
https://www.sciencedirect.com/science/article/pii/S2214860418310492
"Z-Pinning approach for 3D printing mechanically isotropic materials"

BagelOrb, 2019-10-29:
[image]

AbeFM, 2019-11-04:
> It's called 'Z pinning'. Here's a paper about it:
> https://www.sciencedirect.com/science/article/pii/S2214860418310492

Unfortunately, no, there's a link to a page ABOUT where you can buy it or lease it or rent it but certainly not to read it. And the form to pay for it doesn't even have "I just want to see it" as an option.

Just as soon as CURA's perfect, we should all go down there and tell them that you're not distributing information by renting it out.

Looks like a worthwhile paper. If you have a suggestion on where one might read it, let me know. If I track it down I'll share it here.

BagelOrb, 2019-11-04:
www.sci-hub.tw

Kind regards,
Tim Kuipers
PhD researcher at Delft university of technology
Software engineer at Ultimaker

https://www.sciencedirect.com/science/article/pii/S2214860418310492

AbeFM, 2019-11-05:
OMG awesome thanks
http://sci-hub.tw/https://www.sciencedirect.com/science/article/pii/S2214860418310492