Ultimaker/CuraEngine #1043: CuraEngine + UV LCD (Inkspire)

cederom, 2019-04-20:
Hello, what are the changes to adapt CURA Engine for UV LCD printers like Zortrax Inkspire? Maybe Ultimaker is working on printer like this? :-)

rburema, 2019-04-23:
Sorry, the gcode we output is quite specific for FFF type machines. It'd need a _major_ rewrite to be able to _properly_ do other types of printers (or at least ones that are further from the FFF principle, like SLA type printers).

You might still be able to use the Cura frontend (and certainly Uranium) to make an (G)UI for those, but you'd need a different backend (engine) as things stand now.

nallath, 2019-04-23:
Well, you can get the slices from the engine and just stop there. But that would invalidate a large part of the engine (as that's only a few percent of what the slicer does)

cederom, 2019-04-23:
Thank you @nallath and @rburema :-)

I guess these kind of UV LCD are the way to go - a lot better accuracy of the print than FDM, and the slices could be used just before the toolhead path generation.. so the workflow could be forked at some final point of slicing?

You have already lots of good things implemented highly demanded and in fact considered a standard nowadays, but still not all vendors can provide those things that were here in CuraEngine for years now.. also the Open-Source approach still keeps me towards CuraEngine rather than something I have absolutely no control over..

I know this would be lots of work.. I was wondering if anyone already considered such option maybe did some experiments? :-)

Ghostkeeper, 2019-04-24:
We have not done any experiments with this, nor have I heard of others using CuraEngine for their DLP printers. I also think that CuraEngine would be highly unsuitable for this purpose.

Yes, you could probably re-use some algorithms from the [code that generates the cross sections](https://github.com/Ultimaker/CuraEngine/blob/master/src/slicer.cpp). But all the rest of the things that you need will need to be built from scratch, like generating support or infill and making draining holes and such. The algorithms in CuraEngine for that right now are aimed at FFF and not at all suitable for DLP.