Ultimaker/CuraEngine #1044: Layer end Z move

konkav-js, 2019-04-21:
Hi.

When I am printing a plate with more than one object on it, I hear a "bumping" sound at the end of a layer. It looked to me, that at the end of a layer, the head first moved to X/Y coordinates where the new layer print starts, without Z moves, and in it's way it hits the object, and after this move, it does the Z move. 
So I saved some g-code files, and tried to find this out. Here is a snippet of a g-code around a layer change:

```
...
G1 F2400 X120.4 Y138.869 E134.32202    { this is the last print 
G1 X120.4 Y138.781                       move in the layer }
M204 S800                              { set accel. }
M205 X8 Y8                             { set jerk }
G1 F1800 E131.32202                    { retract }
;MESH:NONMESH
G0 F4200 X94.538 Y130.995              { move to X/Y coordinates } *
;TIME_ELAPSED:153.134905
;LAYER:2
;MESH:18650_case.STL
G0 X94.538 Y130.995 Z0.7               { move to same X/Y, plus Z move }
...
```
For me it looks like the last G0 move in a layer (*) is completely redundant, and removing it would solve my "bumping" problem. 

I don't know, if others are experiencing this problem. When I print in thin layers, it is not so bad, but when I want to make a fast print, and I go with 0.3 layer (0.4 nozzle) it is bad. Maybe it is because of my printer, it's on the cheaper side of the spectrum 😃 

This was like this in 3.6, and now in 4.0 is the same (on Win10, but I think it didn't matter).

rburema, 2019-04-24:
What's the value of 'Z Hop when retracted' (a.k.a. `retraction_hop_enabled` when directly talking to the engine)'? I see that the default is 'false' for a lot of printers (custom-FFF for example).

konkav-js, 2019-04-27:
The Z hop is false, indeed. But, I am talking about a different thing. My "bumping" problem is just a story around, how I stumbled across this. Just slice something in Cura, and look in it, no matter what options are you turning on or off, the layer ends are always the same: 
```
G0 X Y      to the coordinates for the new layer print start
some comments
G0 X Y Z    to the same X and Y coordinates, plus the Z move.
```
I don't think this is a bug, per se, just if someone would already do something around the slicing engine, may took out the `G0 X Y` move. It is completely unnecessarily. Without that line, the head would go to the same position, just rising in the same time.

smartavionics, 2019-04-27:
FYI, this travel move is generated by LayerPlanBuffer::addConnectingTravelMove(). As an experiment, I shall try omitting it and see what happens. I'll report back.

smartavionics, 2019-04-27:
I did that and the results are not really ideal. A travel from the end of one layer to the start of the next still happens but it doesn't use combing or retraction so more work is required than simply omitting the travel at the end of the previous layer. IIRC, this stuff was introduced by Tim when the multithreading was added.

Ghostkeeper, 2019-04-29:
There's a chain of requirements there that makes those results not ideal: The move that contains the new Z coordinate needs to be part of the new layer because it has the new Z coordinate as destination point and the export always takes the Z coordinate of the current layer. So it needs to be in the new layer. However in the new layer (since layers are computed concurrently) there is no information about the collision zones of the previous layer, so it can't know which areas to avoid. So the move that writes the new Z coordinate can't take collisions into account. Instead, we make a move on the previous layer that does take collisions into account.

The limitation is inherent in the framework here. It stems from the requirement that the g-code writer always outputs the Z coordinate of the current layer. Doing otherwise would require us to have an additional coordinate in much of our internal structures which would be detrimental for performance.

The assumption is made though that this move is just as safe to make as the other moves in the layer, like normal travel moves. If this move is a problem, it'd be a problem to print multiple parts anyway.

konkav-js, 2019-04-30:
OK, now it sounds logical to me. I don't use Combing, and Avoid Printed Parts, but if I did, then there would be additional moves, not just a one straight move to the beginning of the next layer. I see why it wouldn't be easy to program this.
I think I will try to make a plugin for OctoPrint, to search for this kind of duplicated moves, and to delete one. Thanks.

closed 2019-04-30 by konkav-js