Ultimaker/CuraEngine #1051: Passing model and support to curaengine

mypeopoly, 2019-05-09:
**Application Version**
curaengine from Cura 3.5.1

**Platform**
Windows 

Hi, how does Cura pass the model + generated support  to curaengine for slicing?  Since Curaengine has different settings for support and for the main model, there has to be a way to identify which is which so the correct print speed is applied.  What would be a good place to learn more about this? Thank you.

Ghostkeeper, 2019-05-09:
Cura only passes the model to CuraEngine for slicing. CuraEngine generates the support.

Communication between Cura's front-end and CuraEngine is done via Protobuf over a local socket, abstracted in the libArcus library. The types of messages that the two can exchange are defined in the [Cura.proto](https://github.com/Ultimaker/CuraEngine/blob/master/Cura.proto) file. In there you'll see the message that is used to send over a [3D model](https://github.com/Ultimaker/CuraEngine/blob/7e087188ca62d1a9a42b57da4573ca8d70bae205/Cura.proto#L25) for CuraEngine to slice and also other things like the settings and such. You'll find that each [slice](https://github.com/Ultimaker/CuraEngine/blob/7e087188ca62d1a9a42b57da4573ca8d70bae205/Cura.proto#L11) has a list of global settings, and some [extruders](https://github.com/Ultimaker/CuraEngine/blob/7e087188ca62d1a9a42b57da4573ca8d70bae205/Cura.proto#L19) which have their own settings. Support settings are configurable per extruder, so they would be among those extruder settings.

The support settings such as "Support Speed" are separate settings with different keys. We don't have a set of settings for support and a set of the same settings for infill and another set for skin, or something. We have just one set of settings for each extruder (but it's a bigger set). We do have a separate set of settings for the global settings and another set of settings for each object (in case the settings were modified with the per-object settings tool).

If you've pre-generated support using an external modelling tool like MeshMixer and want it to be printed as support, you can indicate that in the per-object settings tool. That support object gets sent over to CuraEngine as a mesh like any other, but in its per-object settings will be a setting `support_mesh` that flags it as a mesh that should be printed as being support.

nbn555, 2019-05-12:
Hi,

I'm using command line to slicing using CuraEngine. How can I pass support model?

mypeopoly, 2019-05-12:
We are testing. Based on what @Ghostkeeper  posted, it seems like curaengine accepts multi-objects with individual settings (which is a feature in Cura) so we need to export the main model and supports separately and tag them in the per object setting with correct flag.

Ghostkeeper, 2019-05-13:
> How can I pass support model?

`-l <model_file> -s support_mesh=true`

mypeopoly, 2019-05-15:
@Ghostkeeper   we tested the command and sorted of work.  on one hand, model and support are printing at different speed.  On the other hand, support now disconnects from the model. (see attached pic) We also noticed that the print speed support is only affected by this parameter " speed_support_infill".  We wonder if the gap between support and model has to do with the fact that support is now defined as infill mesh? 
[image]

mypeopoly, 2019-05-16:
below are the cura variables used for slicing.  Perhaps it is other support mesh setting that cause support not connecting?
[Moai 100 100um test.zip](https://github.com/Ultimaker/CuraEngine/files/3185066/Moai.100.100um.test.zip)

Ghostkeeper, 2019-05-16:
The model is going to be printing as support, so including all of the support settings such as support printing speed, support X/Y distance, support Z distance, etc. You'd need to override these settings if you want them changed. However there will only ever be one speed for support, because all support will be merged together into one polygon group. We can't keep them separate due to features like Support Horizontal Expansion.

If you really need to print certain support meshes at different speeds, you'd need to try to emulate it by setting skin and wall thickness to 0. It gets kinda complex then.

To fix the connecting issue, you'd need to reduce `support_top_distance` and `support_xy_distance` to 0.

mypeopoly, 2019-05-17:
setting all those to zero connects to the main model. the overall supports looks different from what was passed so something in curaengine is adding extra supports. I check the support generation is disabled.

In design software
[image]

Sliced
[image]

you can see the support trunks are much thicker and there are horizontal marks that were not presented. We are testing different support parameters and would love to get some insights from.

Ghostkeeper, 2019-05-17:
Ah, yeah I see. You need to set the setting "Drop Down Support Mesh" to False (`support_mesh_drop_down`).

I don't know what causes the bases to become wider. I don't really see that in your screenshots.