Ultimaker/CuraEngine #1053: Print same time

kubo10, 2019-05-10:
**Application Version**

2.7

Hi guys , I have this command ./CuraEngine.exe slice -v -j ..\def\ultimaker2.def.json -s layer_height=0.4 -s material_print_temperature=230 -o "test.gcode" -s center_object=true -s support_enable=true -s mesh_rotation_matrix="[[0.7660444321034694,0.07838625453591003,0.6379902218240532],[0.38687958262887023,0.7364023387685565,-0.5550097152312843],[-0.5133226242707254,0.6719874928568474,0.5337908699629322]]" -s mesh_position_x=-20  -e0 -l "3DBenchy.stl" --next -s center_object=true -s mesh_rotation_matrix="[[1.0,0.0,0.0],[0.0,-4.371139000186241E-8,-0.999999999999999],[-0.0,0.999999999999999,-4.371139000186241E-8]]" -s mesh_position_x=50  -e0 -l "3DBenchy.stl"
I wanna print it same time, but -g doesn´t work. Can you help me :) ? This command make Gcode, where first print one 3Dmodel and after finish first start print another model. 

Thx

Ghostkeeper, 2019-05-13:
Leave out the `--next` and it should work.

kubo10, 2019-05-13:
> Leave out the `--next` and it should work.
Yes my friend, but then doesn't work rotation for models. It's generate one rotation for all models.

Ghostkeeper, 2019-05-13:
Upon encountering the `-l` command it should take the latest value for `mesh_rotation_matrix`:

https://github.com/Ultimaker/CuraEngine/blob/b5b17df3e0fd992429ea84a198a848f02efaa63a/src/communication/CommandLine.cpp#L210

Maybe the behaviour is inconsistent with the rest of the settings there? I don't know, but it still works:

    ./CuraEngine.exe slice -v -j ../def/ultimaker2.def.json -s layer_height=0.4 -s material_print_temperature=230 -o "test.gcode" -s center_object=true -s support_enable=true -e0 -s mesh_rotation_matrix="[[0.7660444321034694,0.07838625453591003,0.6379902218240532],[0.38687958262887023,0.7364023387685565,-0.5550097152312843],[-0.5133226242707254,0.6719874928568474,0.5337908699629322]]" -l "3DBenchy.stl" -s mesh_position_x=-20 -s center_object=true -s mesh_rotation_matrix="[[1,0,0],[0,0,-1],[0,1,0]]" -l "3DBenchy.stl" -s mesh_position_x=50

kubo10, 2019-05-16:
Thanks my mate :+1:

closed 2019-05-17 by Ghostkeeper