Ultimaker/CuraEngine #1058: Feature Request: Spiralize with variable Layer Width

TimoErdmann, 2019-05-27:
Hi,

i want to create a Lampshade with an Image imprinted, similar to a Lithophane. I need to print it with the Spiralize mode so that there is no seam!

Is it possible to print a shell object with variable width with the spiralize mode? My Nozzle is .8 mm, the thinnest part in the Lampshade has .3 mm. 

So i have a hollow shell which is loaded to Cura. But if i enable the spiralize mode it won't generate the spiralized gcode, but a single shell object with changing layer with (with the compensate Outerwall setting enabled). The type is defined as Skin.

If i add a solid object the gcode is correctly generated with a constant moving z-aches. But the Layerwidth has a fixed value and i am not getting the desired countour. Which would be defined by the shell stl.

[image]
 

[UM3_Lampshade_acsp.gcode.txt](https://github.com/Ultimaker/CuraEngine/files/3224672/UM3_Lampshade_acsp.gcode.txt)
[Lampshade_acsp.stl.txt](https://github.com/Ultimaker/CuraEngine/files/3224673/Lampshade_acsp.stl.txt)

Ghostkeeper, 2019-05-28:
You mean line width?

I don't think you'd ever get a good result out of that. Even with a direct drive extruder it'll take up to half a second before the flow rate through the nozzle has adjusted to the new intake through the feeder. When printing at 30mm/s, your nozzle would've moved by 15mm in that half second. So your image gets blurred with a 15mm "kernel" horizontally.

I suggest that you use multiple walls which vary in how far/close they are, and a smaller nozzle size if necessary.

TimoErdmann, 2019-05-29:
Shure i meant line width. 

Yes that may be true, but the is alway moving CCW so the image will be slightly rotated and I think the blur will be miminal. 

But Nevertheless i think it would a nice feature independent from my individual application. Moreover the Lamp shade has only variaton in the x-y-plane and it is not a distinct image. 

Isn't it possible to spiralize a shell and after that modify the lines with the outer wall compensation to fit the limition of the dimensions from the shell.

Ghostkeeper, 2019-05-29:
Maybe it's an idea to try to generate some g-code first before trying to implement it as feature in Cura then. Your use case is getting very small and the amount of work to implement it is big. And if we are not sure that it works then... I mean, did you try with the SVGToolpathReader? That plug-in is able to output different line widths for you, albeit only on the first layer.

TimoErdmann, 2019-05-29:
No i didn't test the SVGToolpathReader, i didn't even know this existed. But the result would be the same as  i have testet the variable linewidth. I have a hollow stl file and set the extrusion width wider than the widest wall width. So lets say the maximum all width is 1 mm so i have set the esxtrusion width to 1.2 mm. It is 20 % wider because the width is ohrtoganally 1 mm but if you slice the object hoizontally the width is bigger because of the slope. The shell of the stl defines the width of the extrusion.(with the setting "Outer Wall Compensation" activated) This works great! But I want to use spiralization with this so that there is no seam. 

So the features are already there. It is not working because the Extrusion type is defined as skin and not "Outer-Wall". For all segments which are the same width as the set Extrusion width are defined as "Outer-Wall" and it will be spiraliced.

Ghostkeeper, 2019-05-31:
> I have a hollow stl file and set the extrusion width wider than the widest wall width.

Huh, I would expect that to have no g-code output since Cura doesn't normally print stuff that is smaller than one wall line width.

TimKrautwald, 2019-05-31:
Hi,

I am working on a project with the same problem (wearables and thin flexible prints)) and I believe, that the use case is not so limited. If you want to print a thin walled object with sloping surfaces that needs to have true constant wall thickness, this is the only way to do it. "Vase Mode" with Slicing in "Surface-Mode" on a solid model delivers a clean toolpath, but the width of the produced part will constantly change. This feature comes in handy for wearables and flexibel prints as you need to rely on a clean single wall print. And I see the point for transparent things.

With properly tuned linear advance I never had any issues, even with really long hotends. The change in extrusion speed is relatively slow in most cases. I am not talking about a sharp structure on the outside of a model by running the extruder with different speeds. Imagine a tilted thin walled tube. It changes the extrusion width really smooth.

I think fixing it may just be a thing of changing the definition (maybe just if the spiralize option is activated) from "skin" to "outer wall" for everything. AFAIK no slicers handles this situation well. It would really help to get rid of the seam, as you can not hide it on such thin prints. It ruins the otherwise consistent appearance of the finished part.

smartavionics, 2019-06-05:
Hi @TimKrautwald , I have just posted Linux and WIndows builds of my cura which now includes an experimental implementation of variable width spiralized walls. You can find my releases at https://www.dropbox.com/sh/s43vqzmi4d2bqe2/AAADdYdSu9iwcKa0Knqgurm4a?dl=0

If you try it out, please provide feedback at https://github.com/smartavionics/CuraEngine/issues/7

TimoErdmann, 2019-06-12:
I am closing this because the use case is very small and the amount of work to implement it is big...

closed 2019-06-12 by TimoErdmann

smartavionics, 2019-06-12:
Err, but haven't I provided this feature in my builds? Once proven useful, I was going to put a PR together so that it can go into Ultimaker's Cura.

TimoErdmann, 2019-06-12:
Hey Mark @smartavionics, 

the last response from me was meant sarcastic. I am sorry that this was misunderstood!! I am amazed by your work and i am really grateful for all you have done! Without your work and fast response this wouldn't have been possible. 

I honor your work. 

Closed: It is solved by @smartavionics in the referenced in the issue above.