Ultimaker/CuraEngine #1059: [CURA 4.1] Sliced print hangs after skirt

csetera, 2019-05-29:
**Application Version**
4.1

**Platform**
Mac OS

**Qt**
Downloaded from Cura's Web site

**PyQt**
Downloaded from Cura's Web site

**Display Driver**
Not sure and believe unrelated

**Steps to Reproduce**
I was attempting to re-slice https://www.thingiverse.com/thing:3495390 to add some Z-hop over printed parts.

**Actual Results**
The resulting GCODE printed the skirt and then it started to "hang".  It would just sit there.

**Expected results**
Generate printable GCODE

**Additional Information**
I will attach working GCODE and failing GCODE.  The working GCODE was generated by 4.0 and did not included Z-Hop.  The failing GCODE is using 4.1 and Z-Hop.  What I think *may* be the issue is the lines of GCODE that look like the following:

G1 F17987547480000 Z0.3

That flow rate just doesn't look like it makes sense to me, but I'm also not a GCODE expert and I'm pretty new to 3D printing in general.

csetera, 2019-05-29:
[Octopus_v5.5_spiral_support_fixed_hangs.gcode.zip](https://github.com/Ultimaker/CuraEngine/files/3230410/Octopus_v5.5_spiral_support_fixed_hangs.gcode.zip)

This is the GCODE that hangs

csetera, 2019-05-29:
[Octopus_v5.5_spiral_support_fixed.gcode.zip](https://github.com/Ultimaker/CuraEngine/files/3230415/Octopus_v5.5_spiral_support_fixed.gcode.zip)

This is the older sliced code that works (without an Z-Hop)

smartavionics, 2019-05-29:
Hello @csetera. Yes, that bad feedrate is causing a lot of breakage. You can get around the problem by explicitly setting a z feedrate either in the printer definition or in the profile (Maximum Z Speed).

Ghostkeeper, 2019-05-29:
Duplicate of https://github.com/Ultimaker/Cura/issues/5753

closed 2019-05-29 by Ghostkeeper