Ultimaker/CuraEngine #1062: Skin expand is harmful to part strength

richfelker, 2019-05-29:
According to the documentation, skin expand is supposed to "allow protruding model elements to have better adhesion to the rest of the model, making it stronger or lighter". However, in my experience it makes them much weaker.

Without skin expand underneath a protruding element, the infill lines inside the protruding element will bond directly to the infill lines within the body beneath. This is most effective with 2D infills where they're immediately above, but also holds to a lesser extent for 3D infills where they're offset slightly.

With skin expand, there is one or more layer of skin in between the large body infill and the protruding element infill, and the protruding element's infill bonds relatively poorly to the skin.

I'm not sure whether this should be considered a functionality bug - it's not clear whether skin expansion can be made less-/non-harmful in this kind of situation - or a documentation issue.

richfelker, 2019-05-29:
On further inspection this might be a broken default. The computed value for skin expansion is the total wall thickness, which would only put the skin under the new wall, not cutting into the infill. However the default value in `fdmprinter.def.json` is 2.8, which is **a lot** larger than the default wall thickness of 0.8. This looks like the cause of cutting off the infill (entirely for small protruding features with diameter smaller than 4 mm) and preventing it from bonding like it should. Indeed, changing the skin expand settings to 0.8 explicitly gives the results I would expect.

Ghostkeeper, 2019-05-31:
The reason why expanding skin makes it stronger in our experience is because this closes small gaps in the skin, allowing the printer to continue printing straight lines in sequence without reversing direction and with proper bridging. In our experience it also works best together with a similar Skin Removal Width.

That "default_value" in fdmprinter.def.json should not be used because it gets overwritten by the "value". Only if you're using CuraEngine in command-line mode should it be using that 2.8, which is indeed too big. The "value" of the setting is set to the actual wall thickness. Perhaps your 2.8 is coming from a different profile?

Ghostkeeper, 2019-05-31:
I've corrected the defaults in fdmprinter.def.json to more sane values if you're using command line slicing. Hopefully that is sufficient then.

closed 2019-05-31 by Ghostkeeper

richfelker, 2019-05-31:
I use command line only, which is certainly why I hit this. Since my above comment I've been getting okay results just changing expand to 0.8, but I see you changed the default for both preshrink and expand to 1.0 in the above commit. Is there a motivation for doing them both? I see it more closely matches what the computed value would be for both settings (although 1.0 vs 0.8).

Ghostkeeper, 2019-05-31:
We've seen best results when the two are similar. If the two setting values are similar, the skin is not adjusted for large shapes and remains only supporting the top surface and closing the bottom surface. Together the two perform a morphological opening operation which removes unnecessary skin along the sides of steep walls.

richfelker, 2019-05-31:
Thanks for explaining! That sounds safe from a standpoint of not impacting the strength/integrity of the print and like it should improve print times.

richfelker, 2019-06-22:
Update: the preshrink does affect part integrity. Given a well of width 0.8 or less between walls, the skin on its floor is getting removed and seems to be getting replaced with a line of width 0.4, which doesn't fill the gap between the walls. The result is holes into the interior (infill region) of the object.

I would guess this affects and has always affected the default configuration if used with the GUI. It looks unsafe to have preshrink greater than line width, and even equal or slightly below might cause holes. Should I open a new bug for this?

Ghostkeeper, 2019-06-26:
Creating those holes will have nothing to do with the line width since the Skin Removal Width is part of [area generation](https://github.com/Ultimaker/CuraEngine/blob/master/docs/generating_areas.md) which has no knowledge of what line width you're using for skin or infill.

I'd expect that a piece of skin that is smaller than the Skin Removal Width will get replaced with infill. So if you have a shape with very thin pillars, you could get infill exposed on the outside there. Such is the danger of adjusting Cura's settings.

If you want to open a bug for this, you need to know what Cura is doing wrong. It says that it'll remove areas that are smaller than this width, and so it does. We can also consider adding warning values to make the setting orange if it's set too high.

richfelker, 2019-06-26:
Before I forget, this looks like it should be a good test piece for demonstrating the problem: https://www.thingiverse.com/thing:11895

If I open a bug, it will not be against the feature (which is behaving as intended) but rather against the default settings, which I think shouldn't make holes in the surface. I'll first try to understand what you're saying about how the nozzle/line width is not relevant and see if I can produce a test case that makes holes even with preshrink of 0.4 or less.