Ultimaker/CuraEngine #1068: [4.1.0] Missing thin walls

zebons, 2019-06-06:
**Application Version**

4.1.0
**Platform**

Windows 64bits
**Qt**

**Steps to Reproduce**
Create a Creality printers and slice with the stl joined 
**Actual Results**
Depending on the line width value, missing walls or some lines of the wall at the left side of the model 
**Expected results**
All walls'd be printed as for ultimaker printer

We get missing thin walls with Cura 4.1.0 (same behaviour with Cura 4.0.0) for printers other tha Ultimaker.
To proof that, a friend of mine do an stl model. This model has walls varying in width from 0.4 to 2.5 steps 0.05.

zebons, 2019-06-06:
My solution doesn't seem to be the best one. I do a test print and if I got all the walls for the print sliced with my modified value, these walls are thicker than they should (2.66mm instead of 2.55mm for the rightest wall). With Cura default value, the leftest wall is missing but all the walls present have the right thickness.
Do you have a better solution to get all the walls with the tight thickness ?

Ghostkeeper, 2019-06-07:
A part with 0.4mm width using a line width of 0.4mm is expected to disappear. This is because the centreline of the contour is offset by 0.2mm, and the centreline contour is then a polygon without area which will disappear.

It works for Ultimaker printers because we found a line width of 7/8ths the nozzle size to work best. So we have 0.35mm line width and the centreline is 0.175mm away from the edge. This leaves the centreline contour to be a polygon with a 0.05mm width which will not disappear.

Of course, if you perform an additional Outer Wall Inset of 0.025mm, then you end up at an inset of 0.2mm again and your polygon disappears. This is all expected behaviour.

In my experience, adjusting the line width works better than adjusting the outer wall inset. Just make the line width a few micrometres smaller and you won't notice a difference in the print.

> I do a test print and if I got all the walls for the print sliced with my modified value, these walls are thicker than they should (2.66mm instead of 2.55mm for the rightest wall).

Printing lines close together will cause them to heat each other up and then sag. Cura has no compensation factor for this (yet). We don't know yet if software is the best solution to this problem either.

Your solution of adjusting the default of the outer wall inset by a few micrometres is not a good solution in my opinion because the same problem will occur again when you have a model that is a few micrometres thinner.

closed 2019-06-07 by Ghostkeeper