Ultimaker/CuraEngine #1071: [Cura 4.1] Gyroid Infill increased / longer print time

stefan123t, 2019-06-09:
**Application Version**
Cura 4.1
**Platform**
Linux Mint 18.3 Sylvia, Linux 4.10.0-36-generic x86-64 
**Display Driver**
n/a
**Steps to Reproduce**
Added four times the same FreeCAD model for a hook and hit slice in Cura 3.6 and 4.1.
I noticed a considerable increase / longer print time in Cura 4.1, i.e. 5:14h instead of 3:42h in Cura 3.6
**Actual Results**
The time specification show roughly the same times or Inner Walls, Outer Walls, Retractions, Skin, Skirt and Travel. The Infill increased from 1:54 (51%) to 3:20 (64%) from Cura 3.6 to Cura 4.1, though the Material Estimation remains at 14.80m
**Expected results**
I would expect the same Infill percentage of 30% for Gyroid infill takes the same amount of time to print, or did I miss a parameter to limit the travel speed during Gyroid infill (to max. 40mm/s) after Cura 3.6 had been releases ?
**Additional Information**
Cura 3.6 Time Specification
[image]

Cura 4.1 Time Estimation
[image]

STL-Model served four times on the build plate
[Fenster Haken 8mm.stl.zip](https://github.com/Ultimaker/CuraEngine/files/3270018/Fenster.Haken.8mm.stl.zip)

@smartavionics I saw a lot of your work in #850 went into Gyroid infill. Would you remember and possibly explain what changed between Cura 3.6 and 4.1 that would qualify for such an increase in print time for the infill with otherwise unchanged Slicer settings ?

stefan123t, 2019-06-09:
Just double checked the results with Cura 4.0 and it has the same figures as Cura 3.6, hence the change was introduced between Cura 4.0 and 4.1.

Here are the settings and the results in Cura 4.0

[image]

[image]

smartavionics, 2019-06-10:
Hello @stefan123t , I really need to see the project (.3mf) file for this. So please do File -> Save and then either rename the .3mf to zip or zip it up and attach to this issue. Thanks.

stefan123t, 2019-06-10:
Hi smartavionics,

thanks for your early response. I skipped the Cura 3.6 version though it is older and the change obviously happened between 4.0 and 4.1.

Find attached the two projects from Cura 4.0 and Cura 4.1 respectively for your reference.
[CCR10_Gyroid_Infill_Cura40-41.zip](https://github.com/Ultimaker/CuraEngine/files/3272233/CCR10_Gyroid_Infill_Cura40-41.zip)

Kind regards,
Stefan

smartavionics, 2019-06-10:
Hi @stefan123t , I think the difference in times can be accounted for by the fact that the ccr10 definition has recently been changed to use a much lower jerk value (was 20, now 8). See https://github.com/Ultimaker/Cura/commit/9a9f58428524ee53bfe04b680cc8e1a23cc4f928. That change was done due to some people having problems with skipped layers.

BTW, 30% gyroid infill is probably more than you need for that part so you can save some time by using a lower density. 20% (or even less) could well be adequate. Also, I recommend using the connect infill lines option. Hope this helps.

Ghostkeeper, 2019-06-10:
Setting jerk back to 20 also produces a slicing time of 3h 43m, just one minute off from the original in Cura 4.0. Could be some other minor change there but that's not significant. The time table tooltip then shows that all features print in the same time except that infill still takes 1:55h instead of 1:54.

closed 2019-06-10 by Ghostkeeper

stefan123t, 2019-06-10:
Hi @smartavionics and @Ghostkeeper, thanks for your explanation:
I have a CTC CR10S with a modified Marlin 1.1.9, but did not notice any jerk with the old setting of 20mm/s. I do not understand what jerk has to do with skipped layers, it could affect skipped steps within a layer though. 
I have set Infill Jerk to 20 and it is back to 3:49h total time, or 1:55h infill time. Also I thought that one of the advantages of Gyroid infill is the higher speed you can print it at, so I believe I am fine with the old default.
As for the infill percentage the parts are pretty stiff: I think I maxed the Gyroid infill out with 30%, as your code has some lower limits to reproduce a waveform, it can probably not get much smaller or higher infill % for that matter.
Many thanks for your development and support!

smartavionics, 2019-06-10:
>As for the infill percentage the parts are pretty stiff: I think I maxed the Gyroid infill out with 30%, as your code has some lower limits to reproduce a waveform, it can probably not get much smaller or higher infill % for that matter.

Hmm, really?

Ghostkeeper, 2019-06-11:
> I do not understand what jerk has to do with skipped layers, it could affect skipped steps within a layer though.

Yeah he meant that it causes a layer shift, not a skipped layer.